prominenceai/deepstream-services-library #891: Extent the RTSP Source to support JPEG encoding

rjhowell44, 2022-12-26:
The current RTSP Source implementation only supports H264 and H265 formats with the `decodebin` plugin.  We want to extend the Source component to support JPEG format with the `nvv4l2decoder` plugin. The selection of the depay, parser, and decoder, plugins will be handled in the `select-stream` callback function.

This Issue supersedes Questions about whether rtsp to jpeg decode is supported #884

rjhowell44, 2022-12-26:
Merged into v0.26.alpha

youngjae-avikus, 2023-04-10:
@rjhowell44

Something seems to be tested only on dgpu at this time

When I tried to do it in jetson, the following issues arose,

https://forums.developer.nvidia.com/t/problems-extending-rtsp-decode-source-bin-to-support-jpeg-encoding/249257/5

In DeepStream 6.2, if `mjpeg=1` is set as the nvv4l2decoder option as shown below, the Decoder is producing too many buffer warning does not appear. However, dgpu does not have an `mjpeg` option, and it seems to be automatically decoded.

```bash
gst-launch-1.0 rtspsrc location=rtsp://192.168.0.40:554/jpeg latency=100 ! rtpjpegdepay! jpegparse !
nvv4l2decoder mjpeg=1 ! nvvideoconvert! video/x-raw,format=RGBA ! videoconvert! autovideosink
```

closed 2023-05-03 by rjhowell44