prominenceai/deepstream-services-library #898: a bug in the DeepSort Tracker related tracker_confidence

youngjae-avikus, 2023-01-04:
Hello
https://github.com/prominenceai/deepstream-services-library/pull/879
I'm using an updated tracker api and using monitor-action as an action.

The test was conducted with NvDCF and DeepSort, and the former continues to change tracker-confidence value from monitor-action. The latter, on the other hand, remains at 1. This happens all when `interval >=0` 
Could this be considered an nvidia deepsort library problem?
Thank U

youngjae-avikus, 2023-02-23:
@rjhowell44 

https://forums.developer.nvidia.com/t/is-there-a-way-to-use-only-re-id-score-in-deepsort-tracker/238492/6

they said,
If the tracker algorithm does not generate confidence value, then tracker confidence value will be set to the default value (i.e., 1.0 ) for tracked objects. For IOU, NvSORT and NvDeepSORT trackers, tracker_confidence is set to 1.0 as these algorithms do not generate confidence values for tracked objects. NvDCF tracker, on the other hand, generates confidence for the tracked objects due to its visual tracking capability, and its value is set in tracker_confidence field in NvDsObjectMeta structure.

[https://docs.nvidia.com/metropolis/deepstream/dev-guide/text/DS_plugin_gst-nvtracker.html# 1](https://docs.nvidia.com/metropolis/deepstream/dev-guide/text/DS_plugin_gst-nvtracker.html#)

Then I don't know exactly how to decide whether or not to track the deepsort.

rjhowell44, 2023-02-26:
@youngjae-avikus I'm not sure that I understand your question.  Perhaps you could explain more?  I'm not sure if this helps, but the v0.26.alpha branch has trigger filters for min and max tracker confidence...

[dsl_ode_trigger_tracker_confidence_min_get](https://github.com/prominenceai/deepstream-services-library/blob/v0.26.alpha/docs/api-ode-trigger.md#dsl_ode_trigger_tracker_confidence_min_get)
[dsl_ode_trigger_tracker_confidence_min_set](https://github.com/prominenceai/deepstream-services-library/blob/v0.26.alpha/docs/api-ode-trigger.md#dsl_ode_trigger_tracker_confidence_min_set)
[dsl_ode_trigger_tracker_confidence_max_get](https://github.com/prominenceai/deepstream-services-library/blob/v0.26.alpha/docs/api-ode-trigger.md#dsl_ode_trigger_tracker_confidence_max_get)
[dsl_ode_trigger_tracker_confidence_max_set](https://github.com/prominenceai/deepstream-services-library/blob/v0.26.alpha/docs/api-ode-trigger.md#dsl_ode_trigger_tracker_confidence_max_set)

closed 2024-02-02 by youngjae-avikus