prominenceai/deepstream-services-library #905: Is it possible to set up a makefile build after selecting each function?

youngjae-avikus, 2023-01-29:
Hello
I'm using Src as all files for the makefile I build when I use dsl.
Is there a plan to separate it by function? Because from my point of view, there are several source codes that I don't use (or someone else), so if I turn a tool like Reverse engineering, I get a lot of warning. So, I suggest that it would be good to subdivide the source code so that it can be built by function if I think about it from a lightweight or more service perspective.

youngjae-avikus, 2023-02-09:
I remind you once again,

rjhowell44, 2023-02-09:
@youngjae-avikus this would be far too much effort with almost no return on the investment i.e it does nothing to expand on functionality, or improve performance, security, or quality, etc..  Perhaps something to consider in the future.

closed 2023-02-09 by rjhowell44

rjhowell44, 2023-02-22:
@youngjae-avikus make sure that you are using a release version of DSL when testing your code with specific tools like this ... just as you do with you other dependencies like GStreamer.   I'm sure you're using only a fraction of the GStreamer lib's functionality, however, you're still using the full lib.