prominenceai/deepstream-services-library #930: [request] Implement dsl_sink_encode_dimensions_get/set services (for File, Record, RTSP, WebRTC SInks)

rjhowell44, 2023-03-13:
Need to be able to scale the output of all Encode-Sinks; File, Smart-Record, RTSP, and WebRTC.  This missing functionality was brought up by "trdzung" on discord. 

Important Note: the Smart-Record sink has its own get/set dimension services as provided by the NVIDIA smart record-bin.  The new services set/get the dimensions of the video-converter which is linked upstream of the encoder.

New services
```c
/**
 * @brief Gets the dimensions, width and height, in use by the Encode Sink's
 * input video-converter.
 * @param[in] name name of the Encode Sink to query
 * @param[out] width current width setting in pixels. 0 = no scaling (default).
 * @param[out] height current height setting in pixels. 0 = no scaling (default).
 * @return DSL_RESULT_SUCCESS on success, DSL_RESULT_TILER_RESULT
 */
DslReturnType dsl_sink_encode_dimensions_get(const wchar_t* name, 
    uint* width, uint* height);

/**
 * @brief Sets the dimensions, width and height, for the Encode Sink's input 
 * video converter to use. Set to zero for no-scaling.
 * @param[in] name name of the Record Sink to update
 * @param[in] width new width setting to use in pixels
 * @param[in] height new height setting to use in pixels
 * @return DSL_RESULT_SUCCESS on success, DSL_RESULT_SINK_RESULT
 */
DslReturnType dsl_sink_encode_dimensions_set(const wchar_t* name, 
    uint width, uint height);
```

rjhowell44, 2023-03-13:
Complete and merged into **v0.26.alpha**

BTrDung, 2023-03-29:
Thanks for your support. I will try it as soon as possible.