prominenceai/deepstream-services-library #936: In DeepStream 6.2 Preproc example, problems that are not inference at all

youngjae-avikus, 2023-03-29:
- purpose 
   I am testing DSL in DeepStream 6.2 for advanced nvtracker use. The inference + tracker was not a problem, but there was problem with adding `preprocess` plugin
- Issue Contents
  https://github.com/prominenceai/deepstream-services library/blob/v0.26.alpha/examples/cpp/1file_preproc_pgie_iou_osd_window.cpp
  If I execute this code, there will be no inference result at all, and a blank box will appear like the image attached below

[image]

youngjae-avikus, 2023-03-29:
log 

```txt
root@yyj-avikus:/workspace/aiboat/aiboat/bin# ./dsl-app > log.txt
0:00:00.008663767 11800 0x562380bc2f50 INFO                     DSL DslServices.cpp:136:GetServices:  : Services Initialization
0:00:00.008705227 11800 0x562380bc2f50 INFO                     DSL DslServicesInfo.cpp:58:InfoInitDebugSettings:  : Pre-start log level  = 3,DSL:4
0:00:00.008710876 11800 0x562380bc2f50 INFO                     DSL DslServicesInfo.cpp:59:InfoInitDebugSettings:  : Pre-start Log file  = 
0:00:00.008804767 11800 0x562380bc2f50 INFO                     DSL DslServices.cpp:154:GetServices:  : DSL Version: v0.26.alpha
0:00:00.009567422 11800 0x562380bc2f50 INFO                     DSL DslServices.cpp:165:GetServices:  : Libcurl Initialized Successfully
0:00:00.009576355 11800 0x562380bc2f50 INFO                     DSL DslServices.cpp:166:GetServices:  : Version: 7.68.0
0:00:00.009579443 11800 0x562380bc2f50 INFO                     DSL DslServices.cpp:167:GetServices:  : Host: x86_64-pc-linux-gnu
0:00:00.009585738 11800 0x562380bc2f50 INFO                     DSL DslServices.cpp:168:GetServices:  : Features: 12568477
0:00:00.009588321 11800 0x562380bc2f50 INFO                     DSL DslServices.cpp:169:GetServices:  : SSL Version: OpenSSL/1.1.1f
0:00:00.009590805 11800 0x562380bc2f50 INFO                     DSL DslServices.cpp:170:GetServices:  : Libz Version: 1.2.11
0:00:00.009594587 11800 0x562380bc2f50 INFO                     DSL DslServices.cpp:171:GetServices:  : Protocols: 0x7f574aa85100
0:00:00.011025219 11800 0x562380bc2f50 INFO                     DSL DslSourceBintr.cpp:1472:SetFileUri:  : File Path = file:/opt/nvidia/deepstream/deepstream-6.2/samples/streams/sample_1080p_h265.mp4
0:00:00.012930330 11800 0x562380bc2f50 INFO                     DSL DslAvFile.cpp:79:AvInputFile:  : Video codec data found in media file: /opt/nvidia/deepstream/deepstream/samples/streams/sample_1080p_h265.mp4
0:00:00.012939131 11800 0x562380bc2f50 INFO                     DSL DslAvFile.cpp:80:AvInputFile:  :   dimensions : 1920x1080
0:00:00.012942304 11800 0x562380bc2f50 INFO                     DSL DslAvFile.cpp:81:AvInputFile:  :   frame-rate : 30/1
0:00:00.012965771 11800 0x562380bc2f50 INFO                     DSL DslSourceBintr.cpp:1439:SetUri:  : URI Path for File Source 'uri-source-1' = file:/opt/nvidia/deepstream/deepstream-6.2/samples/streams/sample_1080p_h265.mp4
0:00:00.012984963 11800 0x562380bc2f50 INFO                     DSL DslSourceBintr.cpp:1387:UriSourceBintr:  : 
0:00:00.012990744 11800 0x562380bc2f50 INFO                     DSL DslSourceBintr.cpp:1388:UriSourceBintr:  : Initial property values for UriSourceBintr 'uri-source-1'
0:00:00.012993656 11800 0x562380bc2f50 INFO                     DSL DslSourceBintr.cpp:1389:UriSourceBintr:  :   uri                 : file:/opt/nvidia/deepstream/deepstream-6.2/samples/streams/sample_1080p_h265.mp4
0:00:00.012997209 11800 0x562380bc2f50 INFO                     DSL DslSourceBintr.cpp:1390:UriSourceBintr:  :   is-live             : 0
0:00:00.012999763 11800 0x562380bc2f50 INFO                     DSL DslSourceBintr.cpp:1391:UriSourceBintr:  :   skip-frames         : 0
0:00:00.013001990 11800 0x562380bc2f50 INFO                     DSL DslSourceBintr.cpp:1392:UriSourceBintr:  :   drop-frame-interval : 0
0:00:00.013004442 11800 0x562380bc2f50 INFO                     DSL DslSourceBintr.cpp:1393:UriSourceBintr:  :   width               : 1920
0:00:00.013006893 11800 0x562380bc2f50 INFO                     DSL DslSourceBintr.cpp:1394:UriSourceBintr:  :   height              : 1080
0:00:00.013009186 11800 0x562380bc2f50 INFO                     DSL DslSourceBintr.cpp:1395:UriSourceBintr:  :   fps-n               : 30
0:00:00.013011507 11800 0x562380bc2f50 INFO                     DSL DslSourceBintr.cpp:1396:UriSourceBintr:  :   fps-d               : 1
0:00:00.013014009 11800 0x562380bc2f50 INFO                     DSL DslSourceBintr.cpp:1397:UriSourceBintr:  :   media-out           : video/x-raw(memory:NVMM)
0:00:00.013016361 11800 0x562380bc2f50 INFO                     DSL DslSourceBintr.cpp:1398:UriSourceBintr:  :   buffer-out          : 
0:00:00.013018889 11800 0x562380bc2f50 INFO                     DSL DslSourceBintr.cpp:1399:UriSourceBintr:  :     format            : I420
0:00:00.013021129 11800 0x562380bc2f50 INFO                     DSL DslSourceBintr.cpp:1400:UriSourceBintr:  :     width             : 0
0:00:00.013024479 11800 0x562380bc2f50 INFO                     DSL DslSourceBintr.cpp:1401:UriSourceBintr:  :     height            : 0
0:00:00.013027090 11800 0x562380bc2f50 INFO                     DSL DslSourceBintr.cpp:1402:UriSourceBintr:  :     crop-pre-conv     : 0:0:0:0
0:00:00.013030096 11800 0x562380bc2f50 INFO                     DSL DslSourceBintr.cpp:1403:UriSourceBintr:  :     crop-post-conv    : 0:0:0:0
0:00:00.013033019 11800 0x562380bc2f50 INFO                     DSL DslSourceBintr.cpp:1404:UriSourceBintr:  :     orientation       : 0
0:00:00.013054553 11800 0x562380bc2f50 INFO                     DSL DslServicesSource.cpp:850:SourceFileNew:  : New File Source 'uri-source-1' created successfully
0:00:00.013067551 11800 0x562380bc2f50 INFO                     DSL DslServicesPreproc.cpp:51:PreprocNew:  : Preprocessor config file: /opt/nvidia/deepstream/deepstream/sources/apps/sample_apps/deepstream-preprocess-test/config_preprocess.txt
[... cut ...]
0:00:02.292355233 11893 0x7fb770076b00 FIXME              h265parse gsth265parse.c:1850:gst_h265_parse_parse_frame:<h265parse0> Implement timestamp/duration interpolation based on SEI message
0:00:02.292382279 11893 0x7fb770076b00 FIXME              h265parse gsth265parse.c:1850:gst_h265_parse_parse_frame:<h265parse0> Implement timestamp/duration interpolation based on SEI message
0:00:02.292406066 11893 0x7fb770076b00 FIXME              h265parse gsth265parse.c:1850:gst_h265_parse_parse_frame:<h265parse0> Implement timestamp/duration interpolation based on SEI message
```

youngjae-avikus, 2023-03-29:
By the way, the example of nvidia below works well

```bash
gst-launch-1.0 filesrc location = /opt/nvidia/deepstream/deepstream/samples/streams/sample_1080p_h264.mp4 ! qtdemux ! h264parse ! nvv4l2decoder ! m.sink_0 nvstreammux name=m batch-size=1 width=1920 height=1080 ! nvvideoconvert ! nvdspreprocess config-file= /opt/nvidia/deepstream/deepstream/sources/gst-plugins/gst-nvdspreprocess/config_preprocess.txt  ! nvinfer config-file-path= /opt/nvidia/deepstream/deepstream/samples/configs/deepstream-app/config_infer_primary.txt input-tensor-meta=1 batch-size=7  ! nvmultistreamtiler width=1920 height=1080 ! nvvideoconvert ! nvdsosd ! nveglglessink
```

youngjae-avikus, 2023-03-29:
@rjhowell44 

/opt/nvidia/deepstream/deepstream/sources/apps/sample_apps/deepstream-preprocess-test/deepstream_preprocess_test.cpp

This example works well, but is it possible to check the DSL example because the box does not appear? The preproc, infer  config is almost the same as the model size, so I don't think this part is a problem, but I wonder if it's not compatible with 6.2 or if there are any additional parts that need to be set up

youngjae-avikus, 2023-03-30:
6.0.1 docs
https://docs.nvidia.com/metropolis/deepstream/6.0.1/dev-guide/text/DS_plugin_gst-nvinfer.html
6.2 docs
https://docs.nvidia.com/metropolis/deepstream/dev-guide/text/DS_plugin_gst-nvdspreprocess.html

Something has changed since the output, but I'm posting it because I'm suspicious that it's

rjhowell44, 2023-04-03:
@youngjae-avikus sorry, took a little time to figure this out.  Nothing to do with DS 6.2 but with a recent change I made to the default buffer-out format for all sources from `NV12`  to `I420`.  Changing it back to `NV12` fixed the problem.  It seems that the Preprocessor only likes `NV12`. although I did not try with `RGBA`.  This fix has been merged into the v0.26.alpha branch.

closed 2023-04-03 by rjhowell44

youngjae-avikus, 2023-04-03:
@rjhowell44 
Looking at the revised part, it seems that only buffer-out changes, so it is confirmed that it is a problem in the visualize part, so shouldn't the information on how many were caught with osd_pad_sink come out regardless of buffer-out information? When parsing the buffers in the corresponding pph, does it have to be formatted with nv12?

rjhowell44, 2023-04-03:
@youngjae-avikus the preprocessor does not produce tensor-meta for the infer-engine if the buffer format is I420...  The preprocessing does not work.  I'm not sure why, something I will bring up with NVIDIA.  Setting the format back to NV12 as it was before (and as it is with NVIDIA's example) is the solution.  It is not a visualization issue but a preprocessing issue.