prominenceai/deepstream-services-library #938: Proposal for New Postprocessing Algorithm for Front View

youngjae-avikus, 2023-03-30:
- Background of the Issue:
During the addition of preprocess to the system, it was observed that the inference results for the entire frame and Region of Interest (ROI) came out together. As a result, a merge algorithm was required for post-processing. The merge algorithm was borrowed from "sahi" and used as a reference (https://github.com/obss/sahi/blob/26acb891168a6338f2b9c89b2f002ded968d6cb5/sahi/postprocess/combine.py#L181-L301).

  However, it was found that the post-processing algorithm is only suitable for **drone view** images with a similar composition to the images taken by UAV. In the drone view, there are not many overlapping boxes, so the post-processing algorithm is effective. However, in the front view images such as front-facing camera view, there are more overlapping boxes than expected. As a result, the Intersection over Smaller (IOS) metric of two different boxes is frequently below 0.5, which leads to merging of boxes that should not be merged. Consequently, the mAP, the metric of object detection, came out lower than the system without preprocessing.

- Suggested part
The algorithm to be proposed first has the same texture as the preprocess, but it was judged to be more suitable for the front view, and a lab_test was performed, and mAP increased (0.64 -> 0.67) than the existing tiling method (based SAHI) in our validation set. This is not just the Maritime scene I deal with, but the car-like front view will benefit from tiling (preprocessing) while reducing merge issues in post-processing.

   Suggested Changes:
   - The new algorithm is based on the assumption that the Full Inference part is already trustworthy, and the core procedure involves dividing the algorithm into two methods, **GreedyNMMBaseOrder** and **GreedyNMMnoMergeOrder**.
  - Both methods sort the frames, with the Full Frame part at the front of the order. The key difference is that the improved algorithm, GreedyNMMnoMergeOrder, searches for non-full frame boxes first and deletes them if they overlap with the full frame boxes according to the Intersection over Smaller (IOS) standard.
  - Implementation:
  The code will be sent via Discord, and the change is only in the processImpl part. In DeepStream, the frames are divided by ROI index, and the code contains a syntax like if (nbatch[element] != 3), where 3 is the index for Full Frame.
  
Thank You

youngjae-avikus, 2023-04-14:
Is it okay for me to start working first?

youngjae-avikus, 2023-05-23:
@rjhowell44 

hello. This is pph in the form of pgie_src implemented for the algorithm I proposed above.
First of all, I couldn't interpipe, and first of all I focused on the implementation below .
So, I proceeded in the form of inserting two input_sources, and distinguished Full and ROI inferences through batch_id = 0,1

However, At last `nvds_remove_obj_meta_from_frame` and `nvds_add_obj_meta_to_frame` do not seem to work. Can you confirm the use of the api?

```cpp
uint osd_sink_pad_buffer_probe(void* buffer, void* client_data)
{
    GstBuffer* pGstBuffer = (GstBuffer*)buffer;

    NvDsBatchMeta* pBatchMeta = gst_buffer_get_nvds_batch_meta(pGstBuffer);

    int number_of_class = category_name.size();
    /**
     * @brief FrameObjectData is a variable to distinguish between Full Frame and slice inference.
     * batch_id 0 is Full Frame, 1 is Slice
     */
    std::vector<std::vector<std::vector<float>>> FrameObjectData;
    std::vector<std::vector<NvDsObjectMeta*>> ObjectMetaArray;

    FrameObjectData.resize(number_of_class);
    ObjectMetaArray.resize(number_of_class);

    std::vector<ObjectData> Output;
    
    // 
    NvDsFrameMeta* pBatchZeroFrameMeta = NULL;
    NvDsFrameMeta* pBatchOneFrameMeta = NULL;

    // For each frame in the batched meta data
    for (NvDsMetaList* pFrameMetaList = pBatchMeta->frame_meta_list;
         pFrameMetaList; pFrameMetaList = pFrameMetaList->next)
    {
        // Check for valid frame data
        NvDsFrameMeta* pFrameMeta = (NvDsFrameMeta*)(pFrameMetaList->data);
        if (pFrameMeta != nullptr)
        {
            auto frame_number = pFrameMeta->frame_num;
            auto batch_id = pFrameMeta->batch_id;

            // std::cout << "batch_id: " << batch_id << "\n";
            
            if (batch_id == 0) {
                pBatchZeroFrameMeta = pFrameMeta;
            }
            else {
                pBatchOneFrameMeta = pFrameMeta;
            }

            NvDsMetaList* pObjectMetaList = pFrameMeta->obj_meta_list;

            // For each detected object in the frame.
            while (pObjectMetaList)
            {
                // Check for valid object data
                NvDsObjectMeta* pObjectMeta = (NvDsObjectMeta*)(pObjectMetaList->data);

                std::vector<float> object = {
                    pObjectMeta->rect_params.left,
                    pObjectMeta->rect_params.top,
                    pObjectMeta->rect_params.left + pObjectMeta->rect_params.width,
                    pObjectMeta->rect_params.top + pObjectMeta->rect_params.height,
                    pObjectMeta->confidence,
                    static_cast<float>(batch_id)
                };

                FrameObjectData[pObjectMeta->class_id].emplace_back(object);
[... 219 lines cut ...]
                ObjectMetaArray[cls_id][it - keep_to_merge_list.begin()]->rect_params.top = 
                    FrameObjectData[cls_id][it - keep_to_merge_list.begin()][1];
                ObjectMetaArray[cls_id][it - keep_to_merge_list.begin()]->rect_params.width = 
                    FrameObjectData[cls_id][it - keep_to_merge_list.begin()][2] - 
                    FrameObjectData[cls_id][it - keep_to_merge_list.begin()][0];
                ObjectMetaArray[cls_id][it - keep_to_merge_list.begin()]->rect_params.height = 
                    FrameObjectData[cls_id][it - keep_to_merge_list.begin()][3] - 
                    FrameObjectData[cls_id][it - keep_to_merge_list.begin()][1]; 
            }
        }

        for (auto &x: remove) 
        {
            nvds_remove_obj_meta_from_frame(pBatchOneFrameMeta, ObjectMetaArray[cls_id][x]);
        }

        for (auto &x: add) {
            // assert(FrameObjectData[cls_id][x][5] == -1);
            nvds_add_obj_meta_to_frame(pBatchZeroFrameMeta, ObjectMetaArray[cls_id][x], NULL);
        }
    }

    return DSL_PAD_PROBE_OK;
}
```

rjhowell44, 2023-05-23:
@youngjae-avikus when you say you couldn't interpipe... what does that mean???

youngjae-avikus, 2023-05-24:
hello @rjhowell44 
I just uploaded two files in example/cpp to the [issue-938-interpipe-preproc](https://github.com/prominenceai/deepstream-services-library/tree/issue-938-interpipe-preproc) branch.
First, before testing the new_pph, I put a preproc in two Interpipe-srcs, and the Roi is drawn, but it seems that the inference is not inferred only in the Roi corresponding to it at all, but for the entire frame. Can you please check on this issue?

closed 2024-02-02 by youngjae-avikus

jstumpin, 2025-05-16:
> calculateBoxUnion

Hi, could you provide this function? BTW, has the code work for you?

Thanks.

jstumpin, 2025-05-19:
> Hi, could you provide this function? BTW, has the code work for you?

Never mind, codes provided [here](https://github.com/prominenceai/deepstream-services-library/tree/99f96f9a29145fad691a1fc7ea5544929aef9bef/src/nmp) work like a charm! Faster than my Python implementation.

Kudos @YoungjaeDev