prominenceai/deepstream-services-library #941: multi_image_source에서 capture_action을 하는 법

youngjae-avikus, 2023-04-12:
hello
What I'm trying to do with this code is to evaluate cnn model performance.
I am trying to infer pre-named images for each region through multi_image_source api.

What I want to do is to **save the osd for each image inference result**, but even looking at the reference document of capture_new, it is still a bit awkward, so I would like to ask what is wrong with the current code below.

The test environment is a DeepStream 6.2 dGPU

thank you for your help

```cpp
#include <catch2/catch.hpp>
#include "Dsl.h"
#include "DslApi.h"

using namespace std::literals;

static const std::wstring pipeline_name(L"test-pipeline");
static const std::wstring source_name1(L"image-source-1");
static const std::wstring jpeg_file_path_multi(L"./tests/imgs/FLL_sea.%d.jpg");
static const std::wstring ws_boat_jpeg_file_path_multi(L"./tests/imgs/wangsan_boat.%d.jpg");
static const std::wstring ws_buoy_jpeg_file_path_multi(L"./tests/imgs/wangsan_buoy.%d.jpg");
static const std::wstring preprocess_name(L"preprocessor");
static const std::wstring primary_gie_name(L"primary-gie");
static const uint batch_size(4);

static std::wstring preproc_config_file(
    L"nas_data/FLL/config_preprocess_B4.txt");
static std::wstring infer_config_file(
    L"nas_data/FLL/config_infer_primary_yoloV5.txt");
static std::wstring model_engine_file(
    L"nas_data/FLL/model_b4_gpu0_fp16.engine");

static const uint interval(0);
static const std::wstring tiler_name1(L"tiler");
static const uint tiler_width(1920);
static const uint tiler_height(1080);

static const std::wstring osd_name(L"on-screen-display");
static const boolean text_enabled(true);
static const boolean clock_enabled(false);
static const boolean bbox_enabled(true);
static const boolean mask_enabled(false);

static const uint offest_x(100);
static const uint offest_y(140);
static const uint sink_width(1280);
static const uint sink_height(720);

static const std::wstring window_sink_name(L"window-sink");

static const std::wstring ode_handler_name(L"ode-handler");

static const std::wstring occurrence_trigger_name(L"occurrence-trigger");
static const std::wstring summation_trigger_name(L"summation-trigger");

static const std::wstring offset_label_action_name(L"offset-label-action");
static const std::wstring print_action_name(L"print-action");
static const std::wstring capture_frame_action_name(L"frame-capture-action");

static const std::wstring pipeline_graph_name(L"pipeline-playing");
static uint fps_n(1);
static uint fps_d(1);

static int x_offset(0);
static int y_offset(-25);

//
// Function to be called on End-of-Stream (EOS) event
//
static void eos_event_listener(void* client_data)
{
    std::wcout << L"EOS event for Pipeline " << std::endl;
    std::this_thread::sleep_for(3s);
    dsl_main_loop_quit();
}

//
// Function to be called on Object Capture (and file-save) complete
//
static void capture_complete_listener(dsl_capture_info* info_ptr,
                                      void* client_data)
{
    std::cout << "***  Object Capture Complete  ***" << std::endl;

    std::wcout << L"capture_id: " << info_ptr->capture_id << std::endl;
    std::wcout << L"filename:   " << info_ptr->filename << std::endl;
    std::wcout << L"dirpath:    " << info_ptr->dirpath << std::endl;
    std::wcout << L"width:      " << info_ptr->width << std::endl;
    std::wcout << L"height:     " << info_ptr->height << std::endl;
}

void setup_pipeline_components(const std::wstring& source_file_path)
{
    REQUIRE(dsl_component_list_size() == 0);

    REQUIRE(dsl_source_image_multi_new(source_name1.c_str(),
                                       source_file_path.c_str(), fps_n, fps_d) == DSL_RESULT_SUCCESS);

    REQUIRE(dsl_preproc_new(preprocess_name.c_str(), preproc_config_file.c_str()) == DSL_RESULT_SUCCESS);

    REQUIRE(dsl_infer_gie_primary_new(primary_gie_name.c_str(),
                                      infer_config_file.c_str(), model_engine_file.c_str(),
                                      interval) == DSL_RESULT_SUCCESS);

    // **** IMPORTANT! for best performace we explicity set the GIE's batch-size
    // to the number of ROI's defined in the Preprocessor configuraton file.
    REQUIRE(dsl_infer_batch_size_set(primary_gie_name.c_str(), batch_size) == DSL_RESULT_SUCCESS);

    // **** IMPORTANT! we must set the input-meta-tensor setting to true when
    // using the preprocessor, otherwise the GIE will use its own preprocessor.
    REQUIRE(dsl_infer_gie_tensor_meta_settings_set(primary_gie_name.c_str(), true, false) == DSL_RESULT_SUCCESS);

    REQUIRE(dsl_sink_window_new(window_sink_name.c_str(),
                                offest_x, offest_y, sink_width, sink_height) == DSL_RESULT_SUCCESS);

    REQUIRE(dsl_tiler_new(tiler_name1.c_str(),
                          tiler_width, tiler_height) == DSL_RESULT_SUCCESS);

    REQUIRE(dsl_osd_new(osd_name.c_str(), text_enabled, clock_enabled,
                        bbox_enabled, mask_enabled) == DSL_RESULT_SUCCESS);

    REQUIRE(dsl_ode_action_print_new(print_action_name.c_str(), false) ==
            DSL_RESULT_SUCCESS);

    REQUIRE(dsl_ode_action_capture_frame_new(capture_frame_action_name.c_str(), L"./capture") ==
            DSL_RESULT_SUCCESS);

    REQUIRE(dsl_ode_trigger_summation_new(summation_trigger_name.c_str(),
                                          DSL_ODE_ANY_SOURCE, DSL_ODE_ANY_CLASS, DSL_ODE_TRIGGER_LIMIT_NONE) ==
            DSL_RESULT_SUCCESS);

    const wchar_t* summation_actions[] = {print_action_name.c_str(), NULL};

    REQUIRE(dsl_ode_trigger_action_add_many(summation_trigger_name.c_str(),
                                            summation_actions) == DSL_RESULT_SUCCESS);

    REQUIRE(dsl_ode_trigger_occurrence_new(occurrence_trigger_name.c_str(),
                                           DSL_ODE_ANY_SOURCE, DSL_ODE_ANY_CLASS, DSL_ODE_TRIGGER_LIMIT_NONE) == DSL_RESULT_SUCCESS);

    REQUIRE(dsl_ode_action_label_offset_new(offset_label_action_name.c_str(), x_offset, y_offset) == DSL_RESULT_SUCCESS);

    REQUIRE(dsl_ode_trigger_action_add(occurrence_trigger_name.c_str(), offset_label_action_name.c_str()) == DSL_RESULT_SUCCESS);

    const wchar_t* triggers[] = {occurrence_trigger_name.c_str(), NULL};

    REQUIRE(dsl_pph_ode_new(ode_handler_name.c_str()) == DSL_RESULT_SUCCESS);

    REQUIRE(dsl_pph_ode_trigger_add_many(ode_handler_name.c_str(), triggers) == DSL_RESULT_SUCCESS);

    REQUIRE(dsl_osd_pph_add(osd_name.c_str(), ode_handler_name.c_str(),
                            DSL_PAD_SINK) == DSL_RESULT_SUCCESS);
}

void assemble_pipeline_and_play()
{
    const wchar_t* components[] = {L"image-source-1", L"preprocessor", L"primary-gie", L"on-screen-display", L"window-sink", NULL};

    WHEN("When the Pipeline is Assembled")
    {
        REQUIRE(dsl_pipeline_new(pipeline_name.c_str()) == DSL_RESULT_SUCCESS);

        REQUIRE(dsl_pipeline_component_add_many(pipeline_name.c_str(),
                                                components) == DSL_RESULT_SUCCESS);

        REQUIRE(dsl_pipeline_eos_listener_add(pipeline_name.c_str(),
                                              eos_event_listener, NULL) == DSL_RESULT_SUCCESS);

        THEN("Pipeline is Able to LinkAll and Play")
        {
            REQUIRE(dsl_pipeline_play(pipeline_name.c_str()) == DSL_RESULT_SUCCESS);
            // dsl_pipeline_dump_to_dot(pipeline_name.c_str(),
            //                          const_cast<wchar_t*>(pipeline_graph_name.c_str()));
            dsl_main_loop_run();
            REQUIRE(dsl_pipeline_stop(pipeline_name.c_str()) == DSL_RESULT_SUCCESS);
            dsl_delete_all();
        }
    }
}

SCENARIO("Evaluation of boat, channel marker preprocess inference performance at the coast of Fort Lauderdale",
         "[TILING-FLL-SEA]")
{
    GIVEN("A Pipeline, Multi-image source, Pre-Processor, Primary GIE, Window Sink")
    {
        setup_pipeline_components(jpeg_file_path_multi);
        assemble_pipeline_and_play();
    }
}

SCENARIO("Evaluation of boat tiling inference performance at the coast of Wangsan",
         "[TILING-WANGSAN-BOAT]")
{
    GIVEN("A Pipeline, Multi-image source, Pre-Processor, Primary GIE, Window Sink")
    {
        setup_pipeline_components(ws_boat_jpeg_file_path_multi);
        assemble_pipeline_and_play();
    }
}

SCENARIO("Evaluation of buoy tiling inference performance at the coast of Wangsan",
         "[TILING-WANGSAN-BUOY]")
{
    GIVEN("A Pipeline, Multi-image source, Pre-Processor, Primary GIE, Window Sink")
    {
        setup_pipeline_components(ws_buoy_jpeg_file_path_multi);
        assemble_pipeline_and_play();
    }
}
```

rjhowell44, 2023-04-13:
@youngjae-avikus I see a couple of issues with the above.  First keep in mind that 
* an ODE Hander can have one or more ODE Triggers
* each ODE Trigger can have one or more ODE Actions. 

so if you want your Hander to have both the `summation-trigger` and `occurrence-trigger`, then your `triggers` variable should be defined as
```c
    const wchar_t* triggers[] = {summation_trigger_name.c_str(),
        occurrence_trigger_name.c_str(), NULL};
```

and if you want to have both the `print-action` and `capture-frame-action` added to the `summation-trigger`, then your `summation_actions` variable should be defined as 
```c
    const wchar_t* summation_actions[] = {print_action_name.c_str(), 
        capture_frame_action_name.c_str(), NULL};
```

Next, if you want the OSD results in you captured image you will need to add the ODE Handler to the `DSL_PAD_SRC` (output pad).  Your code above is adding it to the `DSL_PAD_SINK` (input) pad.

youngjae-avikus, 2023-04-13:
Thank you
I changed the code as you said
However, still no image files are saved in the capture folder.

The error in question is the following error...

```bash
nvbufsurface: mapping of memory type (0) not supported
0:00:19.301710525 25080 0x7f013c01a060 ERROR                    DSL DslOdeAction.cpp:526:HandleOccurrence:  : Destination surface failed to map for Action 'frame-capture-action'
```

```cpp
void setup_pipeline_components(const std::wstring& source_file_path)
{
    REQUIRE(dsl_component_list_size() == 0);

    REQUIRE(dsl_source_image_multi_new(source_name1.c_str(),
                                       source_file_path.c_str(), fps_n, fps_d) == DSL_RESULT_SUCCESS);

    REQUIRE(dsl_preproc_new(preprocess_name.c_str(), preproc_config_file.c_str()) == DSL_RESULT_SUCCESS);

    REQUIRE(dsl_infer_gie_primary_new(primary_gie_name.c_str(),
                                      infer_config_file.c_str(), model_engine_file.c_str(),
                                      interval) == DSL_RESULT_SUCCESS);

    // **** IMPORTANT! for best performace we explicity set the GIE's batch-size
    // to the number of ROI's defined in the Preprocessor configuraton file.
    REQUIRE(dsl_infer_batch_size_set(primary_gie_name.c_str(), batch_size) == DSL_RESULT_SUCCESS);

    // **** IMPORTANT! we must set the input-meta-tensor setting to true when
    // using the preprocessor, otherwise the GIE will use its own preprocessor.
    REQUIRE(dsl_infer_gie_tensor_meta_settings_set(primary_gie_name.c_str(), true, false) == DSL_RESULT_SUCCESS);

    REQUIRE(dsl_sink_window_new(window_sink_name.c_str(),
                                offest_x, offest_y, sink_width, sink_height) == DSL_RESULT_SUCCESS);

    REQUIRE(dsl_tiler_new(tiler_name1.c_str(),
                          tiler_width, tiler_height) == DSL_RESULT_SUCCESS);

    REQUIRE(dsl_osd_new(osd_name.c_str(), text_enabled, clock_enabled,
                        bbox_enabled, mask_enabled) == DSL_RESULT_SUCCESS);

    /// summation_trigger | print_action, capture_frame_action
    REQUIRE(dsl_ode_action_print_new(print_action_name.c_str(), false) ==
            DSL_RESULT_SUCCESS);

    REQUIRE(dsl_ode_action_capture_frame_new(capture_frame_action_name.c_str(), L"./capture") ==
            DSL_RESULT_SUCCESS);

    REQUIRE(dsl_ode_trigger_summation_new(summation_trigger_name.c_str(),
                                          DSL_ODE_ANY_SOURCE, DSL_ODE_ANY_CLASS, DSL_ODE_TRIGGER_LIMIT_NONE) ==
            DSL_RESULT_SUCCESS);

    const wchar_t* summation_actions[] = {print_action_name.c_str(), capture_frame_action_name.c_str(), NULL};

    REQUIRE(dsl_ode_trigger_action_add_many(summation_trigger_name.c_str(),
                                            summation_actions) == DSL_RESULT_SUCCESS);
    
    /// occurrence_trigger | offset_label_action
    REQUIRE(dsl_ode_trigger_occurrence_new(occurrence_trigger_name.c_str(),
                                           DSL_ODE_ANY_SOURCE, DSL_ODE_ANY_CLASS, DSL_ODE_TRIGGER_LIMIT_NONE) == DSL_RESULT_SUCCESS);

    REQUIRE(dsl_ode_action_label_offset_new(offset_label_action_name.c_str(), x_offset, y_offset) == DSL_RESULT_SUCCESS);

    REQUIRE(dsl_ode_trigger_action_add(occurrence_trigger_name.c_str(), offset_label_action_name.c_str()) == DSL_RESULT_SUCCESS);

    const wchar_t* triggers[] = {occurrence_trigger_name.c_str(), summation_trigger_name.c_str(), NULL};

    REQUIRE(dsl_pph_ode_new(ode_handler_name.c_str()) == DSL_RESULT_SUCCESS);

[... 70 lines cut ...]
    Source Id       : 0
    Batch Id        : 0
    Pad Index       : 0
    Frame           : 0
    Width           : 1920
    Heigh           : 1080
  Object Data       : ------------------------
    Occurrences     : 2
  Criteria          : ------------------------
    Class Id        : 2147483647
    Min Infer Conf  : 0
    Min Track Conf  : 0
    Min Frame Count : 1 out of 1
    Min Width       : 0
    Min Height      : 0
    Max Width       : 0
    Max Height      : 0
    Inference       : No

nvbufsurface: mapping of memory type (0) not supported
0:00:19.301710525 25080 0x7f013c01a060 ERROR                    DSL DslOdeAction.cpp:526:HandleOccurrence:  : Destination surface failed to map for Action 'frame-capture-action'
EOS event for Pipeline 
===============================================================================
All tests passed (108 assertions in 8 test cases)
```

youngjae-avikus, 2023-04-13:
And the strange thing is that it is currently being received as `image_multi_source,` but due to the handler below, "EOS event for Pipeline" is called as soon as the source comes out. So it seems that the problem is that the window_sink view does not come out at the 1 second interval I set.

```bash
REQUIRE(dsl_pipeline_eos_listener_add(pipeline_name.c_str(),
                                              eos_event_listener, NULL) == DSL_RESULT_SUCCESS);
```

rjhowell44, 2023-04-13:
@youngjae-avikus I'm trying to reproduce the frame-capture issue on a dGPU setup, but having a few issues with the DS 6.2 install.  

RE: eos  I see in your script

```c
static const std::wstring source_name1(L"image-source-1");
static const std::wstring jpeg_file_path_multi(L"./tests/imgs/FLL_sea.%d.jpg");

    REQUIRE(dsl_source_image_multi_new(source_name1.c_str(),
                                       source_file_path.c_str(), fps_n, fps_d) == DSL_RESULT_SUCCESS);
```

it appears that you are using the singe image path and not the mutli image path.

youngjae-avikus, 2023-04-13:
@rjhowell44 
How is it different? I wrote it after looking at the example below. Can you tell me why the one I wrote is not a mutli image path?

https://github.com/prominenceai/deepstream-services-library/blob/master/docs/api-source.md#dsl_source_image_multi_new

my_images...
[image]

youngjae-avikus, 2023-04-13:
Could it be that motion_jpeg is recognized as image-multi-source?

rjhowell44, 2023-04-13:
@youngjae-avikus sorry, my mistake... your example looks correct.  Would it be possible for you to try on your Jetson device? 

Or can you try running https://github.com/prominenceai/deepstream-services-library/blob/master/docs/api-source.md#dsl_source_image_multi_new on your PC setup?

youngjae-avikus, 2023-04-13:
@rjhowell44 

Is the code below correct for the second request? Here's the situation
1. In the code below, the Image appears every set Fps seconds.
2. However, EOS is not recognized. Even if dsl_player_termination_event_listener_add is removed or inserted, the program does not end normally. Must start with Ctrl + C

```cpp
#include <iostream>
#include <sstream>
#include <fstream>
#include <dirent.h>
#include <regex>

#include "DslApi.h"

static const std::wstring jpeg_file_path_multi(L"./tests/imgs/FLL_sea.%d.jpg");
static uint fps_n(1);
static uint fps_d(1);

int WINDOW_WIDTH = DSL_STREAMMUX_DEFAULT_WIDTH;
int WINDOW_HEIGHT = DSL_STREAMMUX_DEFAULT_HEIGHT;

// ##     
// # Function to be called on XWindow KeyRelease event    
// ##     
void xwindow_key_event_handler(const wchar_t* in_key, void* client_data)
{   
    std::wstring wkey(in_key); 
    std::string key(wkey.begin(), wkey.end());
    std::cout << "key released = " << key << std::endl;
    key = std::toupper(key[0]);
    if(key == "P"){
        dsl_player_pause(L"player");
    } else if (key == "R"){
        dsl_player_play(L"player");   
    } else if (key == "Q"){
        dsl_player_stop(L"player");
        dsl_main_loop_quit();
    }
}

//
// Function to be called on End-of-Stream (EOS) event
//
static void eos_event_listener(void* client_data)
{
    std::wcout << L"EOS event for Pipeline " << std::endl;
    // std::this_thread::sleep_for(3s);
    dsl_main_loop_quit();
}

int main(int argc, char** argv)
{  
    DslReturnType retval;

    // # Since we're not using args, we can Let DSL initialize GST on first call    
    while(true){    
        
        retval = dsl_source_image_multi_new(L"multi-image-source",
                                       jpeg_file_path_multi.c_str(), fps_n, fps_d);
        if (retval != DSL_RESULT_SUCCESS) break;      

        // # New Overlay Sink, 0 x/y offsets and same dimensions as Tiled Display    
        retval = dsl_sink_window_new(L"window-sink", 0, 0, WINDOW_WIDTH, WINDOW_HEIGHT);
        if (retval != DSL_RESULT_SUCCESS) break;

        retval = dsl_player_new(L"player", L"multi-image-source", L"window-sink");
        if (retval != DSL_RESULT_SUCCESS) break;
            
        // # Add the XWindow event handler functions defined above    
        retval = dsl_player_xwindow_key_event_handler_add(L"player", xwindow_key_event_handler, nullptr);    
        if (retval != DSL_RESULT_SUCCESS) break;    
        
        retval = dsl_player_termination_event_listener_add(L"player", eos_event_listener, NULL);
        if (retval != DSL_RESULT_SUCCESS) break; 
        
        // # Play the player    
        retval = dsl_player_play(L"player");
        if (retval != DSL_RESULT_SUCCESS) break;

        dsl_main_loop_run();
        retval = DSL_RESULT_SUCCESS;
        break;
    }

    // # Print out the final result
    std::cout << dsl_return_value_to_string(retval) << std::endl;

    // # Cleanup all DSL/GST resources
    dsl_delete_all();

    std::cout<<"Goodbye!"<<std::endl;  
    return 0;
}
```

rjhowell44, 2023-04-14:
@youngjae-avikus I'm able to reproduce the Player terminate issue you described above.  I see the following in the logs
```
0:00:06.234121288 18163   0x556e418290 INFO                     DSL src/DslPipelineStateMgr.cpp:361:HandleEosMessage:  : EOS message recieved for Pipeline 'player'
0:00:06.234225300 18163   0x556e418290 INFO                     DSL src/DslNodetr.h:759:SetState:  : Changing state to 'NULL' for Bintr 'player'
0:00:06.291992361 18163   0x556e418290 INFO                     DSL src/DslNodetr.h:765:SetState:  : State change completed synchronously for Bintr'player'
```

And the shutdown fails to complete.  I will follow up on this.  It sounds like the multi-image source is working for you with this example, correct?

youngjae-avikus, 2023-04-14:
@rjhowell44 

Yes. the multi-image source is working well, but shutdown fails to complete... ! 

```bash
0:00:06.021385471 28990 0x55572d8d4d00 INFO                     DSL DslPipelineStateMgr.cpp:360:HandleEosMessage:  : EOS message recieved for Pipeline 'player'
0:00:06.021486789 28990 0x55572d8d4d00 INFO                     DSL DslNodetr.h:759:SetState:  : Changing state to 'NULL' for Bintr 'player'
0:00:06.075791231 28990 0x55572d8d4d00 INFO                     DSL DslNodetr.h:765:SetState:  : State change completed synchronously for Bintr'player'
```

rjhowell44, 2023-04-14:
@youngjae-avikus the shutdown issue is only related to the Player and not the Pipeline. 

Also, I'm working on the dGPU image-capture (surface map failure) issue now.

rjhowell44, 2023-04-14:
@youngjae-avikus also, if you're trying to capture every frame to jpeg file you could use the Multi-Image Sink.  See the example https://github.com/prominenceai/deepstream-services-library/blob/v0.26.alpha/examples/python/encode_and_save_frame_to_jpeg_thumbnail_periodically.py

but in your case, set fps_n, fps_d, width, and height to 0  - meaning don't change the frame rate or dimensions and save every frame.

```Python
        retval = dsl_sink_image_multi_new('mulit-image-sink', 
            './frame_%04d.jpg', 0, 0, 0, 0)
        if retval != DSL_RETURN_SUCCESS:
            break
```

Then add the multi-image sink along with your window sink to the pipeline.

youngjae-avikus, 2023-04-15:
@rjhowell44 

Hello. You mentioned that I made some modifications, but there are still some issues with my code.
You said that I have three scenarios, and [TILING-WANGSAN-BOAT] works well at the FPS I set, but the other two scenarios skip immediately. If only execute [TILING-FLL-SEA], only one image is created, and that one image is also drawn as shown below.

- [TILING-FLL-SEA]
- [TILING-WANGSAN-BOAT]
- [TILING-WANGSAN-BUOY]

[image]

```cpp
void setup_pipeline_components(const std::wstring& source_file_path)
{
    REQUIRE(dsl_component_list_size() == 0);

    REQUIRE(dsl_source_image_multi_new(source_name1.c_str(),
                                       source_file_path.c_str(), fps_n, fps_d) == DSL_RESULT_SUCCESS);

    REQUIRE(dsl_preproc_new(preprocess_name.c_str(), preproc_config_file.c_str()) == DSL_RESULT_SUCCESS);

    REQUIRE(dsl_infer_gie_primary_new(primary_gie_name.c_str(),
                                      infer_config_file.c_str(), model_engine_file.c_str(),
                                      interval) == DSL_RESULT_SUCCESS);

    // **** IMPORTANT! for best performace we explicity set the GIE's batch-size
    // to the number of ROI's defined in the Preprocessor configuraton file.
    REQUIRE(dsl_infer_batch_size_set(primary_gie_name.c_str(), batch_size) == DSL_RESULT_SUCCESS);

    // **** IMPORTANT! we must set the input-meta-tensor setting to true when
    // using the preprocessor, otherwise the GIE will use its own preprocessor.
    REQUIRE(dsl_infer_gie_tensor_meta_settings_set(primary_gie_name.c_str(), true, false) == DSL_RESULT_SUCCESS);

    REQUIRE(dsl_sink_window_new(window_sink_name.c_str(),
                                offest_x, offest_y, sink_width, sink_height) == DSL_RESULT_SUCCESS);

    REQUIRE(dsl_sink_image_multi_new(image_multi_sink_name.c_str(), L"./frame_%04d.jpg", 0, 0, 0, 0) == DSL_RESULT_SUCCESS);

    REQUIRE(dsl_tiler_new(tiler_name1.c_str(),
                          tiler_width, tiler_height) == DSL_RESULT_SUCCESS);

    REQUIRE(dsl_osd_new(osd_name.c_str(), text_enabled, clock_enabled,
                        bbox_enabled, mask_enabled) == DSL_RESULT_SUCCESS);

    /// summation_trigger | print_action, capture_frame_action
    REQUIRE(dsl_ode_action_print_new(print_action_name.c_str(), false) ==
            DSL_RESULT_SUCCESS);

    REQUIRE(dsl_ode_action_capture_frame_new(capture_frame_action_name.c_str(), L"./capture") ==
            DSL_RESULT_SUCCESS);

    REQUIRE(dsl_ode_trigger_summation_new(summation_trigger_name.c_str(),
                                          DSL_ODE_ANY_SOURCE, DSL_ODE_ANY_CLASS, DSL_ODE_TRIGGER_LIMIT_NONE) ==
            DSL_RESULT_SUCCESS);

    const wchar_t* summation_actions[] = {print_action_name.c_str(), capture_frame_action_name.c_str(), NULL};

    REQUIRE(dsl_ode_trigger_action_add_many(summation_trigger_name.c_str(),
                                            summation_actions) == DSL_RESULT_SUCCESS);
    
    /// occurrence_trigger | offset_label_action
    REQUIRE(dsl_ode_trigger_occurrence_new(occurrence_trigger_name.c_str(),
                                           DSL_ODE_ANY_SOURCE, DSL_ODE_ANY_CLASS, DSL_ODE_TRIGGER_LIMIT_NONE) == DSL_RESULT_SUCCESS);

    REQUIRE(dsl_ode_action_label_offset_new(offset_label_action_name.c_str(), x_offset, y_offset) == DSL_RESULT_SUCCESS);

    REQUIRE(dsl_ode_trigger_action_add(occurrence_trigger_name.c_str(), offset_label_action_name.c_str()) == DSL_RESULT_SUCCESS);

    /// ode_handler / occurrence_trigger
    const wchar_t* triggers[] = {occurrence_trigger_name.c_str(), NULL};
[... 40 lines cut ...]
        setup_pipeline_components(jpeg_file_path_multi);
        assemble_pipeline_and_play();
    }
}

SCENARIO("Evaluation of boat tiling inference performance at the coast of Wangsan",
         "[TILING-WANGSAN-BOAT]")
{
    GIVEN("A Pipeline, Multi-image source, Pre-Processor, Primary GIE, Window Sink")
    {
        setup_pipeline_components(ws_boat_jpeg_file_path_multi);
        assemble_pipeline_and_play();
    }
}

SCENARIO("Evaluation of buoy tiling inference performance at the coast of Wangsan",
         "[TILING-WANGSAN-BUOY]")
{
    GIVEN("A Pipeline, Multi-image source, Pre-Processor, Primary GIE, Window Sink")
    {
        setup_pipeline_components(ws_buoy_jpeg_file_path_multi);
        assemble_pipeline_and_play();
    }
}
```

rjhowell44, 2023-04-18:
Sorry @youngjae-avikus .. away from my disk again Monday and Tuesday.  I'll be able to look into this as soon as I get back

rjhowell44, 2023-05-02:
@youngjae-avikus I've created the following test-suite based on your example above
```c
#include "catch.hpp"
#include "DslApi.h"
#include "Dsl.h"

using namespace std::literals;

static const std::wstring pipeline_name(L"test-pipeline");
static const std::wstring source_name1(L"image-source-1");
static const std::wstring jpeg_file_path_multi(L"./test/streams/sample_720p.%d.jpg");

//static const std::wstring jpeg_file_path_multi(L"./tests/imgs/FLL_sea.%d.jpg");
static const std::wstring ws_boat_jpeg_file_path_multi(L"./tests/imgs/wangsan_boat.%d.jpg");
static const std::wstring ws_buoy_jpeg_file_path_multi(L"./tests/imgs/wangsan_buoy.%d.jpg");
static const std::wstring preprocess_name(L"preprocessor");
static const std::wstring primary_gie_name(L"primary-gie");
static const uint batch_size(4);

static std::wstring preproc_config_file(
    L"/opt/nvidia/deepstream/deepstream/sources/apps/sample_apps/deepstream-preprocess-test/config_preprocess.txt");
static std::wstring infer_config_file(
    L"/opt/nvidia/deepstream/deepstream/samples/configs/deepstream-app/config_infer_primary.txt");
static std::wstring model_engine_file(
    L"/opt/nvidia/deepstream/deepstream/samples/models/Primary_Detector/resnet10.caffemodel_b8_gpu0_int8.engine");

static const uint interval(0);
static const std::wstring tiler_name1(L"tiler");
static const uint tiler_width(1920);
static const uint tiler_height(1080);

static const std::wstring osd_name(L"on-screen-display");
static const boolean text_enabled(true);
static const boolean clock_enabled(false);
static const boolean bbox_enabled(true);
static const boolean mask_enabled(false);

static const uint offest_x(100);
static const uint offest_y(140);
static const uint sink_width(1280);
static const uint sink_height(720);

static const std::wstring window_sink_name(L"window-sink");
static const std::wstring image_multi_sink_name(L"image-multi-sink");

static const std::wstring ode_handler_name(L"ode-handler");

static const std::wstring occurrence_trigger_name(L"occurrence-trigger");
static const std::wstring summation_trigger_name(L"summation-trigger");

static const std::wstring offset_label_action_name(L"offset-label-action");
static const std::wstring print_action_name(L"print-action");
static const std::wstring capture_frame_action_name(L"frame-capture-action");

static const std::wstring pipeline_graph_name(L"pipeline-playing");
static uint fps_n(1);
static uint fps_d(1);

static int x_offset(0);
static int y_offset(-25);

//
// Function to be called on End-of-Stream (EOS) event
//
static void eos_event_listener(void* client_data)
{
    std::wcout << L"EOS event for Pipeline " << std::endl;
    std::this_thread::sleep_for(3s);
    dsl_main_loop_quit();
}
[... 129 lines cut ...]
        assemble_pipeline_and_play();
    }
}

SCENARIO("Evaluation of buoy tiling inference performance at the coast of Wangsan",
         "[MULTI-IMAGE]")
{
    GIVEN("A Pipeline, Multi-image source, Pre-Processor, Primary GIE, Window Sink")
    {
        setup_pipeline_components(jpeg_file_path_multi);
        assemble_pipeline_and_play();
    }
}

```

I added this to my existing test assets to confirm that both the frame-capture-action and multi-image-sink work with v0.26.alpha.  In my case, I'm using the same set of 5 images for all three test cases just for simplicity. 

I can run the above with `$ ./dsl-test-app [MULIT-IMAGE]`

For the multi-image-sink, keep in mind that the saved images will be overwritten with each subsequent test case.  You would need to use a different file naming for each test case (L"./frame_%04d.jpg" is used for all three).

The capture-actions add a unique timestamp to the file-name so all images are saved. 

I just merged one minor change to the v0.26.alpha branch so please make sure to use the latest.

rjhowell44, 2023-05-04:
@youngjae-avikus can we close this issue or are you still having problems?

youngjae-avikus, 2023-05-28:
@rjhowell44 
wait a minute. I'll check

closed 2024-02-02 by youngjae-avikus