prominenceai/deepstream-services-library #969: get stuck when creating the fourth rtsp-sink

shancaidazf, 2023-06-06:
May I ask if this program can create up to 3 sinks and get stuck when creating the fourth sink, thanks!

rjhowell44, 2023-06-07:
@shancaidazf are you reporting an issue with more than 3 rtsp-sinks? Or are you just asking if DSL can support more than 3?

The description of your issue -- ***"get stuck when creating the fourth rtsp-sink"***" -- sounds like you are reporting a bug?

shancaidazf, 2023-06-08:
Thank you very much for your reply. I want to know how to create multiple rtspsins. When I create more than 3 rtspsinks during testing, the interface will get stuck. Am I using the wrong method? What should I do? Thank you

rjhowell44, 2023-06-09:
@shancaidazf I'm able to create a pipeline with 4 rtsp-sinks and can run it on my Jetson device.  In order for me to help you further I need more information.

What device are you using: Jetson or dGPU?
What version of Deepstream are you using.
How are you defining your rtsp-sinks? What values are you using for `udp_port` and `rtmp_port` for each?

Most importantly I need logs.  Please set the following environment variables.
```
export GST_DEBUG=1,DSL:4
export GST_DEBUG_FILE=./dsllogs.txt
```

Please send me logs for both cases
* 3 rtsp-sinks working
* 4 rtsp-sinks getting stuck

shancaidazf, 2023-06-09:
i am running on RTX3070, DGPU environment. deepstream6.2.  the following is my logs.

shancaidazf, 2023-06-09:
[image]

shancaidazf, 2023-06-09:
this is the full logs. https://github.com/shancaidazf/111/blob/main/DSLLOGS.txt

rjhowell44, 2023-06-09:
@shancaidazf looks like we'll need to bring this up with NVIDIA.  According to the table here https://developer.nvidia.com/video-encode-and-decode-gpu-support-matrix-new you should be able to handle 5 concurrent sessions with your dGPU. 

Can I ask you to get me one more log to share with NVIDIA.  Please set the log level to.  
```
export GST_DEBUG=4,DSL:1
```
They will want to see more of there info and less DSL.

shancaidazf, 2023-06-09:
when i set the log level to  this :  export GST_DEBUG=4,DSL:1     the log file:

shancaidazf, 2023-06-09:
when i set the log level to this: export GST_DEBUG=1,nvv4l2h265enc:4
[DSLLOGS.txt](https://github.com/prominenceai/deepstream-services-library/files/11703072/DSLLOGS.txt)

rjhowell44, 2023-06-12:
@shancaidazf please see https://forums.developer.nvidia.com/t/errors-when-using-more-than-3-nvv4l2h265enc-plugins/256281

and try as suggested... upgrade to latest driver version

rjhowell44, 2023-06-13:
@shancaidazf can you let me know if their suggestion worked so I can close the issue with NVIDIA, thanks

shancaidazf, 2023-06-14:
> @shancaidazf can you let me know if their suggestion worked so I can close the issue with NVIDIA, thanks

I just gave it a try and was surprised to find that he succeeded. Thank you very much！

shancaidazf, 2023-06-14:
firstly,thanks to your help.  howerver  i meet a new problem now.  when i try add a rtsp-source and a url-source to a pipeline and use the tilter , it doesnot work, do you know why?

closed 2023-06-14 by rjhowell44