prominenceai/deepstream-services-library #979: Occur error "Bad file descriptor" in pipeline related interpipe-source, interpipe-sink

youngjae-avikus, 2023-06-14:
```txt
                                          / interpipe src \                                                                               / (src0) nvtracker - nvosd - nvsink, file-sink
camera -> interpipe sink -                          - nvstreammux - nvpreprocess - nvinfer - demux 
                                          \ interpipe src /                                                                               \ (src1) fakesink
```                   

I tried to record the result with a file-sink after inputting an 8-second video to run with the above pipeline.
However, the saved video was only saved for 6 seconds, but when I looked at the log, the following errors occurred. Did you know this?
thank you

```txt
0:00:10.511170765 12423 0x5581f3dfdde0 ERROR          v4l2allocator gstv4l2allocator.c:1389:gst_v4l2_allocator_qbuf:<nvv4l2decoder0:pool:src:allocator> failed queueing buffer 2: Bad file descriptor
0:00:10.511194765 12423 0x5581f3dfdde0 ERROR         v4l2bufferpool gstv4l2bufferpool.c:1424:gst_v4l2_buffer_pool_qbuf:<nvv4l2decoder0:pool:src> could not queue a buffer 2
0:00:10.551215917 12423 0x5581f3dfdde0 ERROR          v4l2allocator gstv4l2allocator.c:1389:gst_v4l2_allocator_qbuf:<nvv4l2decoder0:pool:src:allocator> failed queueing buffer 3: Bad file descriptor
0:00:10.551229848 12423 0x5581f3dfdde0 ERROR         v4l2bufferpool gstv4l2bufferpool.c:1424:gst_v4l2_buffer_pool_qbuf:<nvv4l2decoder0:pool:src> could not queue a buffer 3
0:00:10.551362292 12423 0x5581f2fec870 INFO                     DSL DslPipelineStateMgr.cpp:303:HandleBusWatchMessage:  : Message type:: stream-status
0:00:10.551386673 12423 0x5581f2fec870 INFO                     DSL DslPipelineStateMgr.cpp:303:HandleBusWatchMessage:  : Message type:: element
0:00:10.551398977 12423 0x5581f2fec870 INFO                     DSL DslPipelineStateMgr.cpp:303:HandleBusWatchMessage:  : Message type:: element
0:00:10.591173353 12423 0x5581f3dfdde0 ERROR          v4l2allocator gstv4l2allocator.c:1389:gst_v4l2_allocator_qbuf:<nvv4l2decoder0:pool:src:allocator> failed queueing buffer 4: Bad file descriptor
0:00:10.591192291 12423 0x5581f3dfdde0 ERROR         v4l2bufferpool gstv4l2bufferpool.c:1424:gst_v4l2_buffer_pool_qbuf:<nvv4l2decoder0:pool:src> could not queue a buffer 4
0:00:10.631224369 12423 0x5581f2fec870 INFO                     DSL DslPipelineStateMgr.cpp:303:HandleBusWatchMessage:  : Message type:: element
0:00:10.631246837 12423 0x5581f2fec870 INFO                     DSL DslPipelineStateMgr.cpp:303:HandleBusWatchMessage:  : Message type:: element
0:00:10.631258359 12423 0x5581f2fec870 INFO                     DSL DslPipelineStateMgr.cpp:360:HandleEosMessage:  : EOS message recieved for Pipeline 'pipeline'
Pipeline EOS event
0:00:10.631289209 12423 0x5581f2fec870 INFO                     DSL DslPipelineBintr.cpp:404:Stop:  : dsl_pipeline_stop called from bus-watch-function thread context
0:00:10.631306414 12423 0x5581f2fec870 INFO                     DSL DslNodetr.h:774:SetState:  : Changing state to 'NULL' for Bintr 'pipeline'
[NvMultiObjectTracker] De-initialized
0:00:10.664730794 12423 0x5581f2fec870 INFO                     DSL DslNodetr.h:781:SetState:  : State change completed synchronously for Bintr'pipeline'
0:00:10.664785228 12423 0x5581f2fec870 INFO                     DSL DslPipelineSourcesBintr.cpp:305:UnlinkAll:  : Unlinking sources-bin-nvstreammux from interpipe-source-1
0:00:10.664799557 12423 0x5581f2fec870 INFO                     DSL DslNodetr.h:570:UnlinkFromSinkMuxer:  : Unlinking and releasing requested Sink Pad '0x5581f3e02300' for Bintr 'interpipe-source-1'
0:00:10.664819928 12423 0x5581f2fec870 INFO                     DSL DslNodetr.h:593:UnlinkFromSinkMuxer:  : Bintr 'interpipe-source-1' changed state to NULL successfully
0:00:10.665032181 12423 0x5581f2fec870 INFO                     DSL DslPipelineSourcesBintr.cpp:305:UnlinkAll:  : Unlinking sources-bin-nvstreammux from interpipe-source-2
0:00:10.665042410 12423 0x5581f2fec870 INFO                     DSL DslNodetr.h:570:UnlinkFromSinkMuxer:  : Unlinking and releasing requested Sink Pad '0x7fbab4014bf0' for Bintr 'interpipe-source-2'
0:00:10.665052945 12423 0x5581f2fec870 INFO                     DSL DslNodetr.h:593:UnlinkFromSinkMuxer:  : Bintr 'interpipe-source-2' changed state to NULL successfully
0:00:10.665297346 12423 0x5581f2fec870 INFO                     DSL DslMultiComponentsBintr.cpp:226:UnlinkAll:  : Unlinking demuxer-nvstreamdemux from branch-1
0:00:10.665311828 12423 0x5581f2fec870 INFO                     DSL DslNodetr.h:700:CompleteUnlinkFromSourceTee:  : Unlinking and releasing requested Src Pad '0x7fbab4015090' for Bintr 'branch-1'
0:00:10.665379905 12423 0x5581f2fec870 INFO                     DSL DslMultiComponentsBintr.cpp:226:UnlinkAll:  : Unlinking sinks-bin-tee from file-sink
0:00:10.665391881 12423 0x5581f2fec870 INFO                     DSL DslNodetr.h:700:CompleteUnlinkFromSourceTee:  : Unlinking and releasing requested Src Pad '0x5581f3da75f0' for Bintr 'file-sink'
0:00:10.665461985 12423 0x5581f2fec870 INFO                     DSL DslMultiComponentsBintr.cpp:226:UnlinkAll:  : Unlinking sinks-bin-tee from window-sink
0:00:10.665474211 12423 0x5581f2fec870 INFO                     DSL DslNodetr.h:700:CompleteUnlinkFromSourceTee:  : Unlinking and releasing requested Src Pad '0x5581f3da7850' for Bintr 'window-sink'
0:00:10.665628864 12423 0x5581f2fec870 INFO                     DSL DslMultiComponentsBintr.cpp:226:UnlinkAll:  : Unlinking demuxer-nvstreamdemux from fake-sink
0:00:10.665649297 12423 0x5581f2fec870 INFO                     DSL DslNodetr.h:700:CompleteUnlinkFromSourceTee:  : Unlinking and releasing requested Src Pad '0x7fbab40152e0' for Bintr 'fake-sink'
0:00:10.665696351 12423 0x5581f2fec870 INFO                     DSL DslServicesPipeline.cpp:802:PipelineStop:  : Pipeline 'pipeline' transitioned to a state of NULL successfully
0:00:10.665711720 12423 0x5581f2fec870 INFO                     DSL DslPlayerBintr.cpp:338:Stop:  : PlayerBintr is not linked when called to stop
0:00:10.665736752 12423 0x5581f2fec870 INFO                     DSL DslServices.cpp:251:ReturnValueToString:  : Result = 0 = DSL_RESULT_SUCCESS
DSL_RESULT_SUCCESS
0:00:10.715926459 12423 0x5581f2fec870 INFO                     DSL DslServicesPipeline.cpp:96:PipelineDeleteAll:  : All Pipelines deleted successfully
0:00:10.716052823 12423 0x5581f2fec870 INFO                     DSL DslServicesPlayer.cpp:890:PlayerDeleteAll:  : All Players deleted successfully
0:00:10.716560398 12423 0x5581f2fec870 ERROR          v4l2allocator gstv4l2allocator.c:1389:gst_v4l2_allocator_qbuf:<nvv4l2decoder0:pool:src:allocator> failed queueing buffer 0: Bad file descriptor
0:00:10.716589815 12423 0x5581f2fec870 ERROR         v4l2bufferpool gstv4l2bufferpool.c:1424:gst_v4l2_buffer_pool_qbuf:<nvv4l2decoder0:pool:src> could not queue a buffer 0
0:00:10.716602569 12423 0x5581f2fec870 WARN           v4l2allocator gstv4l2allocator.c:838:gst_v4l2_allocator_stop:<nvv4l2decoder0:pool:src:allocator> error releasing buffers buffers: Bad file descriptor
0:00:10.720675864 12423 0x5581f2fec870 INFO                     DSL DslServicesComponent.cpp:76:ComponentDeleteAll:  : All Components deleted successfully
0:00:10.720767880 12423 0x5581f2fec870 INFO                     DSL DslServicesPph.cpp:504:PphDeleteAll:  : All Pad Probe Handlers deleted successfully
0:00:10.720808120 12423 0x5581f2fec870 INFO                     DSL DslServicesOdeTrigger.cpp:2438:OdeTriggerDeleteAll:  : All ODE Triggers deleted successfully
0:00:10.720834401 12423 0x5581f2fec870 INFO                     DSL DslServicesOdeAction.cpp:1841:OdeActionDeleteAll:  : All ODE Actions deleted successfully
0:00:10.720856303 12423 0x5581f2fec870 INFO                     DSL DslServicesDisplayType.cpp:1157:DisplayTypeDeleteAll:  : All Display Types deleted successfully
0:00:10.720866750 12423 0x5581f2fec870 INFO                     DSL DslServicesMessageBroker.cpp:420:MessageBrokerDeleteAll:  : All Message Brokers deleted successfully
Goodbye!
```

rjhowell44, 2023-06-14:
@youngjae-avikus can you show me how you defined your file-sink?

youngjae-avikus, 2023-06-14:
```cpp
 retval = dsl_sink_file_new(L"file-sink", 
            L"./multi2_tiling_output.mp4", DSL_CODEC_H264, DSL_CONTAINER_MP4, 8000000, 0);
        if (retval != DSL_RESULT_SUCCESS) return retval;
        
        // Create first branch to be added to the demuxer tee
        const wchar_t* branch1_component_names[] = 
        {
            L"tracker", L"osd", L"window-sink", L"file-sink",  NULL
        };
```
here

youngjae-avikus, 2023-06-14:
@rjhowell44 
I tried with h265 encoding and mkv container format in dsl_sink_file_new, but the same error occurs
However, when the interpipe pipeline is excluded, it works without problems.

rjhowell44, 2023-06-15:
@youngjae-avikus I'm going to need to try and reproduce this issue -- with out DSL (gstreamer only) -- and post an issue on Ridge Run's repo

youngjae-avikus, 2023-06-15:
@rjhowell44 
I sent you a Discord DM

closed 2024-02-02 by youngjae-avikus