prominenceai/deepstream-services-library #1001: using the 2uri_file_pgie_iou_tracker_demuxer_1osd_1overlay_1window.py

shancaidazf, 2023-07-03:
when there are more than three real-time sources, the video won't play anymore without errors.

shancaidazf, 2023-07-04:
[DSL.txt](https://github.com/prominenceai/deepstream-services-library/files/11943435/DSL.txt)

rjhowell44, 2023-07-04:
@shancaidazf can you share your updated script with me?

shancaidazf, 2023-07-04:
sure

shancaidazf, 2023-07-04:
[2uri_file_pgie_iou_tracker_demuxer_1osd_1overlay_1window.txt](https://github.com/prominenceai/deepstream-services-library/files/11945993/2uri_file_pgie_iou_tracker_demuxer_1osd_1overlay_1window.txt)

rjhowell44, 2023-07-06:
@shancaidazf sorry, yesterday was a holiday here... finally had a chance to run the script above. 

First, I don't have access to your source `uri_h265="http://192.168.101.188:8091/zftest/zftest.live.flv"` but if I chance to one of the file sources, the script runs fine for me.

Looking at your uri it appears you are using a live source??  yet the sources in the script have their `live` property set to false. 
```python
# Two URI File Sources - using the same file.
        retval = dsl_source_uri_new('uri-source-1', uri_h265, False, False, 0)
        if retval != DSL_RETURN_SUCCESS:
            break
        retval = dsl_source_uri_new('uri-source-2', uri_h265, False, False, 0)
        retval = dsl_source_uri_new('uri-source-3', uri_h265, False, False, 0)
```
You will need to set the third parameter to True if using live sources

Next, I see you are creating a branch (branch-2) two hold a fake sink.  If adding a sink to a demuxer, you do not need to create a branch for it.  You only need to create the branch if you have other components with the sink (like branch-1) or if you want to add two sinks to one tee output of the demuxer.

For example
```python
        # New Window Sink, with x/y offsets and dimensions
        retval = dsl_sink_window_new('window-sink', 0, 0, 720, 360)
        retval = dsl_sink_fake_new('fakesink2')
        retval = dsl_sink_fake_new('fakesink3')

        if retval != DSL_RETURN_SUCCESS:
            break

        # New Branch for the PGIE, OSD and Window Sink
        retval = dsl_branch_new_component_add_many('branch1', 
            ['on-screen-display', 'window-sink', None])

        # Add Branch1 and the overlay-sink as Branch2
        retval = dsl_tee_demuxer_new_branch_add_many('demuxer', 
            max_branches=30, branches=['branch1', 'fakesink2', 'fakesink3', None])
```
Last, there was an error in your script. If the called failed it would not exit the script. 

```
        # Add Branch1 and the overlay-sink as Branch2
        retVal = dsl_tee_demuxer_new_branch_add_many('demuxer', 
            max_branches=30, branches=['branch1', 'branch2', None])
        
        if retval != DSL_RETURN_SUCCESS:
            break
```
see `retVal` with upper case `V` and lower case `v`

shancaidazf, 2023-07-06:
Firstly,thanks your reply.  Howerver,  it does  not work as you said.         i  think you can test with this http video file https://vd2.bdstatic.com/mda-pegj3t0z7945rwig/sc/cae_h264/1684491703321969060/mda-pegj3t0z7945rwig.mp4?v_from_s=hkapp-haokan-suzhou&auth_key=1688620894-0-0-8a3db675761716bcc65f026675af6ee1&bcevod_channel=searchbox_feed&cd=0&pd=1&pt=3&logid=3094280804&vid=8281536255871277320&abtest=110930_1&klogid=3094280804

rjhowell44, 2023-07-07:
@shancaidazf  I'm able to reproduce your issue with the uri above. What's interesting is that I have no problem with the following http uri.  http://commondatastorage.googleapis.com/gtv-videos-bucket/sample/ElephantsDream.mp4.  We will need to try and reproduce this with the deepstream_app.

rjhowell44, 2023-07-07:
@shancaidazf I tried with three https sources, tiler, and window-sink, and same issue... so nothing to do with using a demuxer.

I'm able to play two source with your https uri and one source with the http uri as shown below, but adding a fourth source with http fails
[image]

It seems to be more of a decoder/resource limitation.

shancaidazf, 2023-07-07:
But ,  using tiler ,i can run up to 40 streams with the same http url https://vd2.bdstatic.com/mda-pegj3t0z7945rwig/sc/cae_h264/1684491703321969060/mda-pegj3t0z7945rwig.mp4?v_from_s=hkapp-haokan-suzhou&auth_key=1688620894-0-0-8a3db675761716bcc65f026675af6ee1&bcevod_channel=searchbox_feed&cd=0&pd=1&pt=3&logid=3094280804&vid=8281536255871277320&abtest=110930_1&klogid=3094280804

shancaidazf, 2023-07-10:
> @shancaidazf I tried with three https sources, tiler, and window-sink, and same issue... so nothing to do with using a demuxer.
>
> I'm able to play two source with your https uri and one source with [...]

hello,  can you give me more information , why tiler runs well, thansk