prominenceai/deepstream-services-library #1004: Move XWindow services from Pipeline to Window Sink - support multiple Window Sinks per Pipeline.

rjhowell44, 2023-07-10:
Adding the XWindow services to the Pipeline (done during the early days of DSL development) was always a mistake and needs to be corrected.  It was done this way originally because 
1. the Pipeline needs to own the `bus-sync-manager` 
2. it is the `bus-sync-manger` that receives the `prepare-window-handle` message with a pointer to the `nveglglessink` element in that needs the handle.  
3. The Pipeline is unaware of which component (Window Sink) owns the element.

But having the Pipeline/bus-sync-manger create/own the XWindow meant that a Pipeline could only support one Window Sink.

A new scheme will need to be implemented to allow the WindowSinkBintr -- that owns the `nveglglessink` element that is messaging -- to create an own its own XWindow when the message is received by the Pipeline

This will affect almost all examples and many of the behavioral test cases as most use a Window Sink.  

### Breaking changes for the client will be as follows:

The following Pipeline services will be removed
```c
* dsl_pipeline_xwindow_dimensions_get
* dsl_pipeline_xwindow_dimensions_set
* dsl_pipeline_xwindow_handle_get
* dsl_pipeline_xwindow_handle_set
* dsl_pipeline_xwindow_destroy
* dsl_pipeline_xwindow_key_event_handler_add
* dsl_pipeline_xwindow_key_event_handler_remove
* dsl_pipeline_xwindow_button_event_handler_add
* dsl_pipeline_xwindow_button_event_handler_remove
* dsl_pipeline_xwindow_delete_event_handler_add
* dsl_pipeline_xwindow_delete_event_handler_remove
* dsl_pipeline_xwindow_fullscreen_enabled_get
* dsl_pipeline_xwindow_fullscreen_enabled_set
```
The following Player services will be removed
```c
* dsl_player_xwindow_handle_get
* dsl_player_xwindow_handle_set
* dsl_player_xwindow_key_event_handler_add
* dsl_player_xwindow_key_event_handler_remove
```
The following callbacks will be renamed from
```c
* dsl_xwindow_key_event_handler_cb
* dsl_xwindow_button_event_handler_cb
* dsl_xwindow_delete_event_handler_cb
```
to
```c
* dsl_sink_window_key_event_handler_cb
* dsl_sink_window_button_event_handler_cb
* dsl_sink_window_delete_event_handler_cb
```

New Window Sink services replacing the above
```c
/**
 * @brief Gets the current full-screen-enabled setting for the named Window Sink
 * @param[in] name name of the Window Sink to query
 * @param[out] enabled true if full-screen-mode is currently enabled, false otherwise 
  * @return DSL_RESULT_SUCCESS on successful query, DSL_RESULT_SINK_RESULT otherwise.
*/
DslReturnType dsl_sink_window_fullscreen_enabled_get(const wchar_t* name, 
    boolean* enabled);

/**
 * @brief Sets the full-screen-enabled setting for the named Window Sink to use.
 * @param[in] name name of the Window Sink to update.
 * @param[in] enabled if true, sets the XWindow to full-screen on creation.
 * The service will fail if called after the XWindow has been created.
 * @return DSL_RESULT_SUCCESS on successful update, DSL_RESULT_SINK_RESULT otherwise.
 */
DslReturnType dsl_sink_window_fullscreen_enabled_set(const wchar_t* name, 
    boolean enabled);

/**
 * @brief adds a callback to a named Window Sink to be notified on Window 
 * KeyRelease events.
 * @param[in] name name of the Window Sink to update
 * @param[in] handler pointer to the client's function to handle Window key events.
 * @param[in] client_data pointer to client data passed back to the handler function.
 * @return DSL_RESULT_SUCCESS on successful update, DSL_RESULT_SINK_RESULT otherwise.
 */
DslReturnType dsl_sink_window_key_event_handler_add(const wchar_t* name, 
    dsl_sink_window_key_event_handler_cb handler, void* client_data);

/**
 * @brief Removes a callback from a named Window Sink previously added with 
 * dsl_sink_window_key_event_handler_add.
 * @param[in] name name of the Window Sink to update.
 * @param[in] handler pointer to the client's function to remove.
 * @return DSL_RESULT_SUCCESS on successful update, DSL_RESULT_SINK_RESULT otherwise.
 */
DslReturnType dsl_sink_window_key_event_handler_remove(const wchar_t* name, 
    dsl_sink_window_key_event_handler_cb handler);

/**
 * @brief Adds a callback to a named Window Sink be notified on Window 
 * ButtonPress Event
 * @param[in] name name of the Window Sink to update
 * @param[in] handler pointer to the client's function to handle Window button events.
 * @param[in] client_data pointer to client data passed back to the handler function.
 * @return DSL_RESULT_SUCCESS on successful update, DSL_RESULT_SINK_RESULT otherwise.
 */
DslReturnType dsl_sink_window_button_event_handler_add(const wchar_t* name, 
    dsl_sink_window_button_event_handler_cb handler, void* client_data);

/**
 * @brief Removes a callback from a named Window Sink previously added with 
 * dsl_sink_window_button_event_handler_add
 * @param[in] name name of the Window Sink to update
 * @param[in] handler pointer to the client's function to remove
 * @return DSL_RESULT_SUCCESS on successful update, DSL_RESULT_SINK_RESULT otherwise.
 */ 
DslReturnType dsl_sink_window_button_event_handler_remove(const wchar_t* name, 
    dsl_sink_window_button_event_handler_cb handler);

/**
 * @brief Adds a callback to a named Window Sink to be notified on Window 
 * Delete message event
 * @param[in] name name of the Window Sink to update
 * @param[in] handler pointer to the client's function to handle a Window Delete event.
 * @param[in] client_data pointer to client data passed back to the handler function.
 * @return DSL_RESULT_SUCCESS on successful update, DSL_RESULT_SINK_RESULT otherwise.
 */
DslReturnType dsl_sink_window_delete_event_handler_add(const wchar_t* name, 
    dsl_sink_window_delete_event_handler_cb handler, void* client_data);

/**
 * @brief removes a callback from a named Window Sink previously added with 
 * dsl_sink_window_delete_event_handler_add
 * @param[in] name name of the Window Sink to update
 * @param[in] handler pointer to the client's function to remove
 * @return DSL_RESULT_SUCCESS on successful update, DSL_RESULT_SINK_RESULT otherwise.
 */
DslReturnType dsl_sink_window_delete_event_handler_remove(const wchar_t* name, 
    dsl_sink_window_delete_event_handler_cb handler);
```

rjhowell44, 2023-07-10:
This work has been merged into the `v0.27.alpha` release branch.

youngjae-avikus, 2023-07-13:
@rjhowell44 
I don't quite understand, but you mean I can create multiple window-sinks in the demux by building it?

rjhowell44, 2023-07-13:
@youngjae-avikus it means each branch, either from demuxer tee or splitter tee, can have a Window Sink.

closed 2023-09-13 by rjhowell44