prominenceai/deepstream-services-library #1005: Error 'queuing buffer 4: Bad file descriptor error occurred" while doing 4 file_sinks of 4 sources with demux

youngjae-avikus, 2023-07-11:
Hello, the code I tested is as follows

https://github.com/prominenceai/deepstream-services-library/blob/log_error/examples/cpp/4file_file_sink.cpp

The approximate code structure is as follows
```bash
                                                       / tiler - osd - window-sink                                
4-video -> nvstreammux -  splitter -               
                                                       \ demux - file-sink 1,2,3,4                              
```

The error comes out as below\
```bash
1688:HandleSourceElementOnPadAdded:  : Video decode linked for URI source 'file-source-4'
0:00:00.250382602  6947 0x559d5b7b6b60 ERROR          v4l2allocator gstv4l2allocator.c:784:gst_v4l2_allocator_start:<file-sink-4-nvv4l2h264enc:pool:src:allocator> error requesting 2 buffers: Invalid argument
0:00:00.250471893  6947 0x559d5b7b6b60 ERROR         v4l2bufferpool gstv4l2bufferpool.c:1185:gst_v4l2_buffer_pool_start:<file-sink-4-nvv4l2h264enc:pool:src> we received 0 buffer from device '/dev/v4l2-nvenc', we want at least 2
0:00:00.250482354  6947 0x559d5b7b6b60 ERROR             bufferpool gstbufferpool.c:559:gst_buffer_pool_set_active:<file-sink-4-nvv4l2h264enc:pool:src> start failed
0:00:00.252305655  6947 0x559d5b7b3870 ERROR                    DSL src/DslNodetr.h:956:SetState:  : FAILURE occured waiting for state to change to 'PAUSED' for GstNotetr 'pipeline'
0:00:00.252318671  6947 0x559d5b7b3870 ERROR                    DSL src/DslPipelineBintr.cpp:323:Play:  : Failed to Pause non-live soures before playing Pipeline 'pipeline'
0:00:00.252350214  6947 0x559d5b7b3870 INFO                     DSL src/DslServices.cpp:251:ReturnValueToString:  : Result = 524301 = DSL_RESULT_PIPELINE_FAILED_TO_PLAY
DSL_RESULT_PIPELINE_FAILED_TO_PLAY
0:00:00.252741515  6947 0x559d5b7b3870 WARN                     DSL src/DslPipelineBintr.cpp:472:HandleStop:  : Pipeline 'pipeline' failed to receive final EOS message on dsl_pipeline_stop
0:00:00.252748076  6947 0x559d5b7b3870 INFO                     DSL src/DslNodetr.h:924:SetState:  : Changing state to 'NULL' for GstNotetr 'pipeline'
0:00:00.252774032  6947 0x559d5b7b3870 INFO                     DSL src/DslNodetr.h:932:SetState:  : State change completed synchronously for GstNotetr'pipeline'
0:00:00.252816265  6947 0x559d5b7b3870 INFO                     DSL src/DslPipelineSourcesBintr.cpp:320:UnlinkAll:  : Unlinking sources-bin-nvstreammux from file-source-1
0:00:00.305335268  6947 0x559d5b7b3870 INFO                     DSL src/DslNodetr.h:636:UnlinkFromSinkMuxer:  : GstNotetr 'file-source-1' changed state to NULL successfully
0:00:00.305402846  6947 0x559d5b7b3870 INFO                     DSL src/DslNodetr.h:667:UnlinkFromSinkMuxer:  : Unlinking and releasing requested sink pad '0x559d5c047770' for GstNotetr 'file-source-1'
0:00:00.308432435  6947 0x559d5b7b3870 INFO                     DSL src/DslPipelineSourcesBintr.cpp:320:UnlinkAll:  : Unlinking sources-bin-nvstreammux from file-source-2
0:00:00.313698365  6947 0x559d5b7b3870 INFO                     DSL src/DslNodetr.h:636:UnlinkFromSinkMuxer:  : GstNotetr 'file-source-2' changed state to NULL successfully
0:00:00.313749901  6947 0x559d5b7b3870 INFO                     DSL src/DslNodetr.h:667:UnlinkFromSinkMuxer:  : Unlinking and releasing requested sink pad '0x559d5c0479c0' for GstNotetr 'file-source-2'
0:00:00.345957486  6947 0x7fe3e4060360 ERROR          v4l2allocator gstv4l2allocator.c:1389:gst_v4l2_allocator_qbuf:<nvv4l2decoder2:pool:src:allocator> failed queueing buffer 4: Bad file descriptor
0:00:00.345980403  6947 0x7fe3e4060360 ERROR         v4l2bufferpool gstv4l2bufferpool.c:1424:gst_v4l2_buffer_pool_qbuf:<nvv4l2decoder2:pool:src> could not queue a buffer 4
```

The Dockerfile I used is below

```docker
FROM nvcr.io/nvidia/deepstream:6.2-devel

ENV DEBIAN_FRONTEND=noninteractive
ENV LANG C.UTF-8
ENV PATH="/usr/local/cuda/bin:${PATH}"
ENV LD_LIBRARY_PATH="/opt/nvidia/deepstream/deepstream/lib:/usr/local/cuda/lib64:${LD_LIBRARY_PATH}"

# public gpg key error issue
# https://github.com/NVIDIA/nvidia-docker/issues/1631#issuecomment-1112682423

# RUN apt-key del 7fa2af80

RUN apt-get update && apt-get remove --autoremove -y \
    libgstreamer1.0*

RUN	apt-get update && \
    apt-get install -y \
    libopencv-dev \
    software-properties-common \
    libgstrtspserver-1.0-dev \
    gstreamer1.0-rtsp \
    gstreamer1.0-tools \
    libapr1 \
    libapr1-dev \
    libaprutil1 \
    libaprutil1-dev \
    libgeos-dev \
    libcurl4-openssl-dev \
    libjson-glib-1.0-0 \
    libsoup-gnome2.4-dev \
    libgstreamer-plugins* \
    gstreamer1.0-plugins-base \
    gstreamer1.0-plugins-good \
    gstreamer1.0-plugins-bad \
    gstreamer1.0-plugins-ugly

# Install gst-interpipe
WORKDIR /tmp

RUN apt-get install -y \
    libgstreamer1.0-dev \
    libgstreamer-plugins-base1.0-dev \
    gtk-doc-tools

RUN git clone --recursive https://github.com/RidgeRun/gst-interpipe.git

WORKDIR gst-interpipe

RUN ./autogen.sh --libdir /usr/lib/x86_64-linux-gnu/ && \
    make -j16 && \
    make check && \
    make install

# Install gst-interpipe
RUN apt-get install -y

COPY nvdspreprocess_solve_interval.patch /
RUN patch -p0 /opt/nvidia/deepstream/deepstream/sources/gst-plugins/gst-nvinfer/gstnvinfer.cpp < /nvdspreprocess_solve_interval.patch

WORKDIR /opt/nvidia/deepstream/deepstream/sources/gst-plugins/gst-nvinfer

RUN CUDA_VER=11.8 make -j4
RUN CUDA_VER=11.8 make install

WORKDIR /workspace

# RUN gst-inspect-1.0 interpipe

# RUN ldconfig

RUN update-alternatives --install /usr/bin/python python /usr/bin/python3.8 1

RUN python -m pip install --upgrade pip
RUN python -m pip install numpy opencv-python

WORKDIR /opt

ARG version=1.1.6
ENV URL="https://github.com/NVIDIA-AI-IOT/deepstream_python_apps/releases/download/v$version/pyds-$version-py3-none-linux_x86_64.whl"

RUN apt-get update && apt-get install -y wget python3.8-dev

RUN wget $URL \
    && pip install pyds-$version-py3-none-linux_x86_64.whl

# Install User package
RUN apt-get update && \
    apt-get install -y \
    vim \ 
    silversearcher-ag
```

rjhowell44, 2023-07-12:
@youngjae-avikus I'm able to run your example on a Jetson platform... I have not had a chance to try on dGPU yet

Your issue seems very similar to another user's issue, which we logged on NVIDIA's forum here. https://forums.developer.nvidia.com/t/errors-when-using-more-than-3-nvv4l2h265enc-plugins/256281/6

The solution was to reinstall the latest driver.

youngjae-avikus, 2023-07-12:
> @youngjae-avikus I'm able to run your example on a Jetson platform... I have not had a chance to try on dGPU yet
>
> Your issue seems very similar to another user's issue, which we logged on NVIDIA's [...]

By driver, do you mean Nvidia driver?
My driver version is 12.0 (525.125.06)
Can you tell me which link is in the issue for the user you mentioned?

youngjae-avikus, 2023-07-13:
I changed the driver version to 12.2 and it works fine.

closed 2023-07-13 by youngjae-avikus