prominenceai/deepstream-services-library #1009: process all the videos in the folder

HappyKerry, 2023-07-17:
I want to process(detect & track) all the videos in the folder.

By using dsl_source_file_new can new the file source for the pipeline, but how to change the pipeline when one file stream ends, and change to process the next video?

HappyKerry, 2023-08-08:
> I want to process(detect & track) all the videos in the folder.
> 
> By using dsl_source_file_new can create a file source for the pipeline, but how to change the pipeline when one file stream ends, and change to process the next video?

https://forums.developer.nvidia.com/t/the-new-nvstreammux-cannot-go-back-to-playing-state-after-eos/262303/3

rjhowell44, 2023-08-09:
@HappyKerry sorry for the delay on this.. I will put together an example of how to do this.  Will probably take me a couple of days to get it done.

HappyKerry, 2023-08-09:
> @HappyKerry sorry for the delay on this.. I will put together an example of how to do this. Will probably take me a couple of days to get it done.

thank you

rjhowell44, 2023-08-14:
@HappyKerry, I've checked in a new example to the v0.27.alpha dev branch that processes all mp4 files in a given folder.

See: [process_all_mp4_files_in_folder.py](https://github.com/prominenceai/deepstream-services-library/blob/v0.27.alpha/examples/python/process_all_mp4_files_in_folder.py).

I would suggest that you upgrade to the latest.  I noticed that the Streammuxer generates a couple of critical errors when relinking the source each time the Pipeline is played, although it still seems to work.
```
(python3:9302): GLib-CRITICAL **: 21:44:46.402: g_hash_table_lookup: assertion 'hash_table != NULL' failed
(python3:9302): GLib-CRITICAL **: 21:44:46.402: g_hash_table_insert_internal: assertion 'hash_table != NULL' failed
```
I have not had a chance to try with the latest release(s) and new Streammuxer

HappyKerry, 2023-08-15:
> @HappyKerry, I've checked in a new example to the v0.27.alpha dev branch that processes all mp4 files in a given folder.
> 
> See: [process_all_mp4_files_in_folder.py](https://github.com/prominenceai/deepstream-services-library/blob/v0.27.alpha/examples/python/process_all_mp4_files_in_folder.py).
> 
> I would suggest that you upgrade to the latest. I noticed that the Streammuxer generates a couple of critical errors when relinking the source each time the Pipeline is played, although it still seems to work.
> 
> ```
> (python3:9302): GLib-CRITICAL **: 21:44:46.402: g_hash_table_lookup: assertion 'hash_table != NULL' failed
> (python3:9302): GLib-CRITICAL **: 21:44:46.402: g_hash_table_insert_internal: assertion 'hash_table != NULL' failed
> ```
> 
> I have not had a chance to try with the latest release(s) and new Streammuxer. I will need to follow-up on the errors.

thanks, I will debug and test the codes

closed 2023-09-13 by rjhowell44