prominenceai/deepstream-services-library #1010: stuck after add a secondary infer gie

HappyKerry, 2023-07-19:
I add a secondary infer gie in the example ode_cross_polygon_border_print_event.cpp, but after that the program stucked,

rjhowell44, 2023-07-19:
@HappyKerry if you do upgrade, there are a few changes that you will need to make in your code. See issue #1004 
The example `ode_cross_polygon_border_print_event.cpp` has been updated in this branch

HappyKerry, 2023-07-21:
> @HappyKerry I think you might be asking too much from your pipeline/device. From your inference config file, I see that you are on a Jetson device... which one? What is the frame-rate of your recorded videos?
> 
> Having a two GIE's, both with the `inference-interval` set to 0, a DCF Tracker, and a high-frame rate might be causing the issue.
> 
> The first thing I would try is to set the `inference-interval` on your GIEs to something other than 0. Try 4 to start with.
> 
> Second, in the current release branch that I'm working on (v0.27.alpha) I've added the ability to control/adjust the frame-rate. So just as a test, if you move to the latest release, you can reduce the frame-rate into the GIE's buy calling `dsl_source_video_buffer_out_frame_rate_set` to see if you can get your Pipeline to run.
> 
> Let's first determine if this is a gpu/cpu limitation or perhaps some other issue with DSL.

I am tesing on tesla v100,not on jetson,  I changed the code to v0.27.alpha, stuck also happened, there might some bugs here

HappyKerry, 2023-07-21:
> @HappyKerry I think you might be asking too much from your pipeline/device. From your inference config file, I see that you are on a Jetson device... which one? What is the frame-rate of your [...]

I am tesing on tesla v100,not on jetson,  I changed the code to v0.27.alpha, stuck also happened, there might some bugs here

rjhowell44, 2023-07-21:
@HappyKerry please try with inference-interval = 4 for both gies ... the results of this will be very helpful

HappyKerry, 2023-07-24:
> @HappyKerry please try with inference-interval = 4 for both gies ... the results of this will be very helpful

i try with inference-interval=4 for both gies, stuck also hanppened

rjhowell44, 2023-07-24:
@HappyKerry thanks for the above, it is very helpful. 

First, the fact that you can run in deepstream (I'm assuming you mean the deepstream_app or some other example) means we will be able to get it running is DSL. 

Second, we have dozens of test cases, examples, and users (that I know of) that are using multiple gies, so your general use case is nothing new... so I need to understand your specific use case in more detail. 

Looking at your log file, I don't see anything out of the ordinary, except for the final statement.
```
DSL src/DslPipelineStateMgr.cpp:325:HandleBusWatchMessage:�[00m  : Unhandled message type:: latency
```
Unfortunately, we don't know which plugin is reporting the latency issue in the current release (something that will be addressed in the next release). 

All that said, there are a couple of quick tests I would like you to try just to get more data.  
- can you try replacing the file-sink with a window-sink, to see if it still gets stuck.
- can you try setting the `sync` property for the file-sink to false. see [dsl_sink_sync_enabled_set](https://github.com/prominenceai/deepstream-services-library/blob/master/docs/api-sink.md#dsl_sink_sync_enabled_set)

Beyond that, the easiest way to move forward, without a bunch of back and forth, would be to produce a Pipeline graph from both the deepstream-app pipeline and DSL pipeline... although, I'm more interested in the working case with deepstream.

Let me know if this is something you are willing and can do. Happy to provide details on how to capture if need be.

HappyKerry, 2023-08-01:
> @HappyKerry thanks for the above, it is very helpful.
> 
> First, the fact that you can run in deepstream (I'm assuming you mean the deepstream_app or some other example) means we will be able to get it running is DSL.
> 
> Second, we have dozens of test cases, examples, and users (that I know of) that are using multiple gies, so your general use case is nothing new... so I need to understand your specific use case in more detail.
> 
> Looking at your log file, I don't see anything out of the ordinary, except for the final statement.
> 
> ```
> DSL src/DslPipelineStateMgr.cpp:325:HandleBusWatchMessage:�[00m  : Unhandled message type:: latency
> ```
> 
> Unfortunately, we don't know which plugin is reporting the latency issue in the current release (something that will be addressed in the next release).
> 
> All that said, there are a couple of quick tests I would like you to try just to get more data.
> 
> * can you try replacing the file-sink with a window-sink, to see if it still gets stuck.
> * can you try setting the `sync` property for the file-sink to false. see [dsl_sink_sync_enabled_set](https://github.com/prominenceai/deepstream-services-library/blob/master/docs/api-sink.md#dsl_sink_sync_enabled_set)
> 
> Beyond that, the easiest way to move forward, without a bunch of back and forth, would be to produce a Pipeline graph from both the deepstream-app pipeline and DSL pipeline... although, I'm more interested in the working case with deepstream.
> 
> Let me know if this is something you are willing and can do. Happy to provide details on how to capture if need be.
> 
> P.S. can you tell me which deepstream example and config-file you used?

I set the sync property for the file-sink to false, it is also stucked, but when i remove the secondary gie, the pipeline runs ok

rjhowell44, 2023-08-03:
@HappyKerry please provide me with a full log including a gstreamer/deepstream plugins
```
export GST_DEBUG=4
```

MintyLee, 2023-10-09:
Is this problem solved? I'm also having the same issue

youngjae-avikus, 2023-10-09:
I didn't have any inconveniences with SGIE, but it would be better if you shared the SGIE configuration.

MintyLee, 2023-10-09:
> I didn't have any inconveniences with SGIE, but it would be better if you shared the SGIE configuration.

2uri_file_pgie_iou_tracker_3sgie_tiler_osd_custom_pph_window.py I ran this example, the configuration files are the default ones

rjhowell44, 2023-10-09:
@MintyLee please provide logs... 
```
export GST_DEBUG=1,DSL:4
```
Please provide details on your Platform (dGPU or Jetson), DS Version, etc.

MintyLee, 2023-10-11:
> @MintyLee please provide logs...
> 
> ```
> export GST_DEBUG=1,DSL:4
> ```
> 
> Please provide details on your Platform (dGPU or Jetson), DS Version, etc.

Platform:dGPU, [GeForce RTX 3070], DeepStream6.3(by the way, this issue appears on deepstream6.3, which does not appear on deepstream6.2, but does not rule out other factors),Graphic card driver:525.125.06, deepstream-services-library: 0.27.alpha

The log is displayed as follows:
`0:00:00.006754112 3241738      0x38c78c0 INFO                     DSL src/DslServices.cpp:136:GetServices:  : Services Initialization
0:00:00.006794870 3241738      0x38c78c0 INFO                     DSL src/DslServicesInfo.cpp:58:InfoInitDebugSettings:  : Pre-start log level  = 1,DSL:4
0:00:00.006798117 3241738      0x38c78c0 INFO                     DSL src/DslServicesInfo.cpp:59:InfoInitDebugSettings:  : Pre-start Log file  = 
0:00:00.006970306 3241738      0x38c78c0 INFO                     DSL src/DslServices.cpp:154:GetServices:  : DSL Version: v0.27.alpha
0:00:00.007553124 3241738      0x38c78c0 INFO                     DSL src/DslServices.cpp:165:GetServices:  : Libcurl Initialized Successfully
0:00:00.007557540 3241738      0x38c78c0 INFO                     DSL src/DslServices.cpp:166:GetServices:  : Version: 7.68.0
0:00:00.007560371 3241738      0x38c78c0 INFO                     DSL src/DslServices.cpp:167:GetServices:  : Host: x86_64-pc-linux-gnu
0:00:00.007563784 3241738      0x38c78c0 INFO                     DSL src/DslServices.cpp:168:GetServices:  : Features: 12568477
0:00:00.007566220 3241738      0x38c78c0 INFO                     DSL src/DslServices.cpp:169:GetServices:  : SSL Version: OpenSSL/1.1.1f
0:00:00.007568351 3241738      0x38c78c0 INFO                     DSL src/DslServices.cpp:170:GetServices:  : Libz Version: 1.2.11
0:00:00.007571364 3241738      0x38c78c0 INFO                     DSL src/DslServices.cpp:171:GetServices:  : Protocols: 0x7fa788912100
0:00:00.016262197 3241738      0x38c78c0 INFO                     DSL src/DslSourceBintr.cpp:1977:SetFileUri:  : File Path = file:/opt/nvidia/deepstream/deepstream-6.3/samples/streams/sample_1080p_h265.mp4
0:00:00.020243115 3241738      0x38c78c0 INFO                     DSL src/ffmpeg/DslAvFile.cpp:78:AvInputFile:  : Video codec data found in media file: /opt/nvidia/deepstream/deepstream/samples/streams/sample_1080p_h265.mp4
0:00:00.020258586 3241738      0x38c78c0 INFO                     DSL src/ffmpeg/DslAvFile.cpp:79:AvInputFile:  :   dimensions : 1920x1080
0:00:00.020262150 3241738      0x38c78c0 INFO                     DSL src/ffmpeg/DslAvFile.cpp:80:AvInputFile:  :   frame-rate : 30/1
0:00:00.020289740 3241738      0x38c78c0 INFO                     DSL src/DslSourceBintr.cpp:1944:SetUri:  : URI Path for File Source 'uri-h264' = file:/opt/nvidia/deepstream/deepstream-6.3/samples/streams/sample_1080p_h265.mp4
0:00:00.020312372 3241738      0x38c78c0 INFO                     DSL src/DslSourceBintr.cpp:1892:UriSourceBintr:  : 
0:00:00.020315471 3241738      0x38c78c0 INFO                     DSL src/DslSourceBintr.cpp:1893:UriSourceBintr:  : Initial property values for UriSourceBintr 'uri-h264'
0:00:00.020319166 3241738      0x38c78c0 INFO                     DSL src/DslSourceBintr.cpp:1894:UriSourceBintr:  :   uri                 : file:/opt/nvidia/deepstream/deepstream-6.3/samples/streams/sample_1080p_h265.mp4
0:00:00.020322546 3241738      0x38c78c0 INFO                     DSL src/DslSourceBintr.cpp:1895:UriSourceBintr:  :   is-live             : 0
0:00:00.020325152 3241738      0x38c78c0 INFO                     DSL src/DslSourceBintr.cpp:1896:UriSourceBintr:  :   skip-frames         : 0
0:00:00.020327095 3241738      0x38c78c0 INFO                     DSL src/DslSourceBintr.cpp:1897:UriSourceBintr:  :   drop-frame-interval : 0
0:00:00.020330188 3241738      0x38c78c0 INFO                     DSL src/DslSourceBintr.cpp:1898:UriSourceBintr:  :   width               : 1920
0:00:00.020334370 3241738      0x38c78c0 INFO                     DSL src/DslSourceBintr.cpp:1899:UriSourceBintr:  :   height              : 1080
0:00:00.020336400 3241738      0x38c78c0 INFO                     DSL src/DslSourceBintr.cpp:1900:UriSourceBintr:  :   fps-n               : 30
0:00:00.020339566 3241738      0x38c78c0 INFO                     DSL src/DslSourceBintr.cpp:1901:UriSourceBintr:  :   fps-d               : 1
0:00:00.020341670 3241738      0x38c78c0 INFO                     DSL src/DslSourceBintr.cpp:1902:UriSourceBintr:  :   media-out           : video/x-raw(memory:NVMM)
0:00:00.020345684 3241738      0x38c78c0 INFO                     DSL src/DslSourceBintr.cpp:1903:UriSourceBintr:  :   buffer-out          : 
0:00:00.020348542 3241738      0x38c78c0 INFO                     DSL src/DslSourceBintr.cpp:1904:UriSourceBintr:  :     format            : NV12
0:00:00.020351230 3241738      0x38c78c0 INFO                     DSL src/DslSourceBintr.cpp:1905:UriSourceBintr:  :     width             : 0
0:00:00.020354265 3241738      0x38c78c0 INFO                     DSL src/DslSourceBintr.cpp:1906:UriSourceBintr:  :     height            : 0
0:00:00.020356656 3241738      0x38c78c0 INFO                     DSL src/DslSourceBintr.cpp:1907:UriSourceBintr:  :     fps-n             : 0
0:00:00.020358950 3241738      0x38c78c0 INFO                     DSL src/DslSourceBintr.cpp:1908:UriSourceBintr:  :     fps-d             : 0
[... cut ...]
0:00:25.687560612 3241738      0x38c78c0 INFO                     DSL src/DslPipelineStateMgr.cpp:349:HandleBusWatchMessage:  :    info      : Source added, reconfigured nvmultistreamtiler
0:00:25.687572804 3241738      0x38c78c0 INFO                     DSL src/DslPipelineStateMgr.cpp:351:HandleBusWatchMessage:  :    debug     : gstnvtiler.cpp(134): gst_nvmultistreamtiler_sink_event (): /GstPipeline:pipeline/GstBin:tiler/GstNvMultiStreamTiler:tiler-nvmultistreamtiler:
Configuration 1x2`

The code gets stuck every time it runs here

MintyLee, 2023-10-11:
> @MintyLee please provide logs...
> 
> ```
> export GST_DEBUG=1,DSL:4
> ```
> 
> Please provide details on your Platform (dGPU or Jetson), DS Version, etc.

> Platform:dGPU, [GeForce RTX 3070], DeepStream6.3(by the way, this issue appears on deepstream6.3, which does not appear on deepstream6.2, but does not rule out other factors),Graphic card driver:525.125.06, deepstream-services-library: 0.27.alpha

> The log is displayed as follows:

> 
```
0:00:00.006754112 3241738 0x38c78c0 INFO DSL src/DslServices.cpp:136:GetServices: : Services Initialization
0:00:00.006794870 3241738 0x38c78c0 INFO DSL src/DslServicesInfo.cpp:58:InfoInitDebugSettings: : Pre-start log level = 1,DSL:4
0:00:00.006798117 3241738 0x38c78c0 INFO DSL src/DslServicesInfo.cpp:59:InfoInitDebugSettings: : Pre-start Log file =
0:00:00.006970306 3241738 0x38c78c0 INFO DSL src/DslServices.cpp:154:GetServices: : DSL Version: v0.27.alpha
0:00:00.007553124 3241738 0x38c78c0 INFO DSL src/DslServices.cpp:165:GetServices: : Libcurl Initialized Successfully
0:00:00.007557540 3241738 0x38c78c0 INFO DSL src/DslServices.cpp:166:GetServices: : Version: 7.68.0
0:00:00.007560371 3241738 0x38c78c0 INFO DSL src/DslServices.cpp:167:GetServices: : Host: x86_64-pc-linux-gnu
0:00:00.007563784 3241738 0x38c78c0 INFO DSL src/DslServices.cpp:168:GetServices: : Features: 12568477
0:00:00.007566220 3241738 0x38c78c0 INFO DSL src/DslServices.cpp:169:GetServices: : SSL Version: OpenSSL/1.1.1f
0:00:00.007568351 3241738 0x38c78c0 INFO DSL src/DslServices.cpp:170:GetServices: : Libz Version: 1.2.11
0:00:00.007571364 3241738 0x38c78c0 INFO DSL src/DslServices.cpp:171:GetServices: : Protocols: 0x7fa788912100
0:00:00.016262197 3241738 0x38c78c0 INFO DSL src/DslSourceBintr.cpp:1977:SetFileUri: : File Path = file:/opt/nvidia/deepstream/deepstream-6.3/samples/streams/sample_1080p_h265.mp4
0:00:00.020243115 3241738 0x38c78c0 INFO DSL src/ffmpeg/DslAvFile.cpp:78:AvInputFile: : Video codec data found in media file: /opt/nvidia/deepstream/deepstream/samples/streams/sample_1080p_h265.mp4
0:00:00.020258586 3241738 0x38c78c0 INFO DSL src/ffmpeg/DslAvFile.cpp:79:AvInputFile: : dimensions : 1920x1080
0:00:00.020262150 3241738 0x38c78c0 INFO DSL src/ffmpeg/DslAvFile.cpp:80:AvInputFile: : frame-rate : 30/1
0:00:00.020289740 3241738 0x38c78c0 INFO DSL src/DslSourceBintr.cpp:1944:SetUri: : URI Path for File Source 'uri-h264' = file:/opt/nvidia/deepstream/deepstream-6.3/samples/streams/sample_1080p_h265.mp4
0:00:00.020312372 3241738 0x38c78c0 INFO DSL src/DslSourceBintr.cpp:1892:UriSourceBintr: :
0:00:00.020315471 3241738 0x38c78c0 INFO DSL src/DslSourceBintr.cpp:1893:UriSourceBintr: : Initial property values for UriSourceBintr 'uri-h264'
0:00:00.020319166 3241738 0x38c78c0 INFO DSL src/DslSourceBintr.cpp:1894:UriSourceBintr: : uri : file:/opt/nvidia/deepstream/deepstream-6.3/samples/streams/sample_1080p_h265.mp4
0:00:00.020322546 3241738 0x38c78c0 INFO DSL src/DslSourceBintr.cpp:1895:UriSourceBintr: : is-live : 0
0:00:00.020325152 3241738 0x38c78c0 INFO DSL src/DslSourceBintr.cpp:1896:UriSourceBintr: : skip-frames : 0
0:00:00.020327095 3241738 0x38c78c0 INFO DSL src/DslSourceBintr.cpp:1897:UriSourceBintr: : drop-frame-interval : 0
0:00:00.020330188 3241738 0x38c78c0 INFO DSL src/DslSourceBintr.cpp:1898:UriSourceBintr: : width : 1920
0:00:00.020334370 3241738 0x38c78c0 INFO DSL src/DslSourceBintr.cpp:1899:UriSourceBintr: : height : 1080
0:00:00.020336400 3241738 0x38c78c0 INFO DSL src/DslSourceBintr.cpp:1900:UriSourceBintr: : fps-n : 30
0:00:00.020339566 3241738 0x38c78c0 INFO DSL src/DslSourceBintr.cpp:1901:UriSourceBintr: : fps-d : 1
0:00:00.020341670 3241738 0x38c78c0 INFO DSL src/DslSourceBintr.cpp:1902:UriSourceBintr: : media-out : video/x-raw(memory:NVMM)
0:00:00.020345684 3241738 0x38c78c0 INFO DSL src/DslSourceBintr.cpp:1903:UriSourceBintr: : buffer-out :
0:00:00.020348542 3241738 0x38c78c0 INFO DSL src/DslSourceBintr.cpp:1904:UriSourceBintr: : format : NV12
0:00:00.020351230 3241738 0x38c78c0 INFO DSL src/DslSourceBintr.cpp:1905:UriSourceBintr: : width : 0
0:00:00.020354265 3241738 0x38c78c0 INFO DSL src/DslSourceBintr.cpp:1906:UriSourceBintr: : height : 0
0:00:00.020356656 3241738 0x38c78c0 INFO DSL src/DslSourceBintr.cpp:1907:UriSourceBintr: : fps-n : 0
0:00:00.020358950 3241738 0x38c78c0 INFO DSL src/DslSourceBintr.cpp:1908:UriSourceBintr: : fps-d : 0
0:00:00.020361224 3241738 0x38c78c0 INFO DSL src/DslSourceBintr.cpp:1909:UriSourceBintr: : crop-pre-conv : 0:0:0:0
0:00:00.020363594 3241738 0x38c78c0 INFO DSL src/DslSourceBintr.cpp:1910:UriSourceBintr: : crop-post-conv : 0:0:0:0
0:00:00.020365775 3241738 0x38c78c0 INFO DSL src/DslSourceBintr.cpp:1911:UriSourceBintr: : orientation : 0
0:00:00.020410399 3241738 0x38c78c0 INFO DSL src/DslServicesSource.cpp:813:SourceUriNew: : New URI Source 'uri-h264' created successfully
0:00:00.020675505 3241738 0x38c78c0 INFO DSL src/DslSourceBintr.cpp:1977:SetFileUri: : File Path = file:/opt/nvidia/deepstream/deepstream-6.3/samples/streams/sample_1080p_h265.mp4
0:00:00.023465225 3241738 0x38c78c0 INFO DSL src/ffmpeg/DslAvFile.cpp:78:AvInputFile: : Video codec data found in media file: /opt/nvidia/deepstream/deepstream/samples/streams/sample_1080p_h265.mp4
0:00:00.023473071 3241738 0x38c78c0 INFO DSL src/ffmpeg/DslAvFile.cpp:79:AvInputFile: : dimensions : 1920x1080
0:00:00.023476319 3241738 0x38c78c0 INFO DSL src/ffmpeg/DslAvFile.cpp:80:AvInputFile: : frame-rate : 30/1
[... cut ...]
0:00:25.687549104 3241738 0x38c78c0 INFO DSL src/DslPipelineStateMgr.cpp:347:HandleBusWatchMessage: : Message type : info
0:00:25.687560612 3241738 0x38c78c0 INFO DSL src/DslPipelineStateMgr.cpp:349:HandleBusWatchMessage: : info : Source added, reconfigured nvmultistreamtiler
0:00:25.687572804 3241738 0x38c78c0 INFO DSL src/DslPipelineStateMgr.cpp:351:HandleBusWatchMessage: : debug : gstnvtiler.cpp(134): gst_nvmultistreamtiler_sink_event (): /GstPipeline:pipeline/GstBin:tiler/GstNvMultiStreamTiler:tiler-nvmultistreamtiler:
Configuration 1x2
``` 

> The code gets stuck every time it runs here.

rjhowell44, 2023-10-11:
@MintyLee Thank you... I can actually reproduce this.  I will let you know when it's fixed

MintyLee, 2023-10-25:
> @MintyLee Thank you... I can actually reproduce this. I will let you know when it's fixed
Has this problem been located？

rjhowell44, 2023-10-25:
@MintyLee issue #1086 was opened for this, which has the explanation for failure if you're interested... along with issue #1090 .

Both have been closed with a new patch - version `v0.27.a.alpha` - released today.  Please upgrade to the latest... and my apologies for how long this took to resolve.

closed 2023-10-25 by rjhowell44