prominenceai/deepstream-services-library #1021: Can you tell me where program is finally released when I press ctrl+c twice?

youngjae-avikus, 2023-07-24:
Hello, when I press **ctrl+c** only once in a program like the C++ dsl example, return 0 is called well while exiting the `dsl_main_loop_run function.`

But when I press twice in a row, it doesn't come to return 0 and seems to end somewhere. If I want to handle the situation, where can I write it?

rjhowell44, 2023-07-31:
@youngjae-avikus sorry, need to do a little research here and get back to you

rjhowell44, 2023-08-14:
@youngjae-avikus If you take a look at the implementation of `dsl_main_loop_start` here https://github.com/prominenceai/deepstream-services-library/blob/9410945f8d8fadb3e4b2850dd3d97c7cb019b454/src/DslServices.cpp#L71C6-L71C6

DSL installs a sig-interrupt handler for program termination. When called, the handler will simply call `dsl_main_loop_quit` which then reinstalls the default handler... see: https://github.com/prominenceai/deepstream-services-library/blob/9410945f8d8fadb3e4b2850dd3d97c7cb019b454/src/DslServices.cpp#L79

Are you finding a problem with this implementation.

closed 2024-02-02 by youngjae-avikus