prominenceai/deepstream-services-library #1042: docs of dsl_ode_trigger_custom_new is missing parametes

master-LIN777, 2023-08-14:
I quote this constructor in my code to get the meta in buffer ,but the IDE reminds me that the constructor is missing two parametes:"source" and "client_post_processor".
`retval = dsl_ode_trigger_custom_new('my-custom-trigger',
        DSL_ODE_ANY_CLASS, DSL_ODE_TRIGGER_LIMIT_NONE, my_check_for_occurrence_cb, my_client_data)`
so I add my own source into it just like other constructors ,but I did`t find any methods to create a "client_post_processor" callback.
finally I find a declaration of it in the source code of "dsl.py".
`# dsl_ode_check_for_occurrence_cb
DSL_ODE_CHECK_FOR_OCCURRENCE = \
    CFUNCTYPE(c_bool, c_void_p, c_void_p, c_void_p, c_void_p)`
but I still don`t know how to use it ,because the meaning of parametes in this function is unknown to me,maybe you guys should update the docs since you have already update the code.

rjhowell44, 2023-08-14:
@master-LIN777 thanks for bringing this to my attention... and it looks we need an example for using the custom-trigger... give me a couple of days I'll try and take care of this.

rjhowell44, 2023-09-13:
My apologies for the very late response.  After reviewing my notes, I was able to recall what happened with the Trigger.  Unfortunately, I was only able to support this with C++ (not python).  One of only two API that have this issue.  It has to do with the callback parameters  `frame_meta` and `object_meta` which can not by mapped using the NVIDIA python bindings. 

```c
typedef boolean (*dsl_ode_check_for_occurrence_cb)(void* buffer,
    void* frame_meta, void* object_meta, void* client_data);
```
New DSL mappings will need to be created for these types.  This will need to be addressed in a future release.  The documentation has been updated to reflect this.

closed 2024-05-29 by rjhowell44