prominenceai/deepstream-services-library #1052: Problem

githubzhao735, 2023-08-21:
Python example.  Why does my program will be blocked when I use dsl_pipeline_component_remove to remove one source?
The main information like these:
[image]
 
'''
dGPU
Deepstream6.2
32 sources
Random blocking
main-loop running when I remove the source
'''
and the logs like this
[dsl.log](https://github.com/prominenceai/deepstream-services-library/files/12394644/dsl.log)
Thanks

rjhowell44, 2023-08-22:
@githubzhao735 thanks for the log... on first glance, it looks like you are really pushing DSL!   This is great!!!  Any help on in-service testing is greatly appreciated. 

Here's some initial observations.  At `0:00:00.361635194` I see that the first RTSP-Source  'rtsp-source' links up the video stream.
```
0:00:00.361635194 [332m 7117[00m 0x7fdd88019d20 [36mINFO   [00m [00m                 DSL src/DslSourceBintr.cpp:3100:HandleSourceElementOnPadAdded:[00m  : rtspsrc element linked for RtspSourceBintr 'rtsp-source'
```
next, at `0:00:10.037014756` the second source is created.
```
0:00:10.037014756 [332m 7117[00m      0x1cb6130 [36mINFO   [00m [00m                 DSL src/DslServicesSource.cpp:1525:SourceRtspNew:[00m  : New RTSP Source 'source-2' created successfully
```
at `0:00:10.104048829` the second source `source-2` has linked up the video stream
```
0:00:10.104048829 [332m 7117[00m 0x7fdd4002b800 [36mINFO   [00m [00m                 DSL src/DslSourceBintr.cpp:3100:HandleSourceElementOnPadAdded:[00m  : rtspsrc element linked for RtspSourceBintr 'source-2'
```
at `0:00:12.721861667` I see the first _**error**_ occurring for `source-2`
```
0:00:12.721861667 [332m 7117[00m 0x7fdda803a090 [31;01mERROR  [00m [00m                 DSL src/DslPipelineStateMgr.cpp:412:HandleErrorMessage:[00m  : Error message 'The stream is in the wrong format.' received from 'source-2-rtph264depay'
```
Directly after this, I see the application trying to remove `source-2`
```
0:00:15.062301645 [332m 7117[00m 0x7fdd28002b60 [36mINFO   [00m [00m                 DSL src/DslPipelineSourcesBintr.cpp:236:RemoveChild:[00m  : Unlinking sources-bin-nvstreammux from source-2
```
And the unlinking fails... 
```
0:00:15.088445301 [332m 7117[00m 0x7fdd28002b60 [31;01mERROR  [00m [00m                 DSL src/DslNodetr.h:628:UnlinkFromSinkMuxer:[00m  : Bintr 'source-2' failed to unlink from Sink Muxer
0:00:15.088507568 [332m 7117[00m 0x7fdd28002b60 [31;01mERROR  [00m [00m                 DSL src/DslPipelineSourcesBintr.cpp:244:RemoveChild:[00m  : PipelineSourcesBintr 'sources-bin' failed to Unlink Child Source 'source-2'
```
At this point (approximately 20% of the way into the log) the Pipeline is in a bad state!!!

We need to understand 
1. why is the video stream in a bad format??? and 
2. why does the source fail to unlink from the Streammuxer once the error is reported??

Just FYI, I've already fixed one issue related to dynamic source removal in the current, ongoing [v0.27.alpha dev branch](https://github.com/prominenceai/deepstream-services-library/tree/v0.27.alpha). 

...see **PipelineMultiSourcesBintr is not clearing pad-id for reuse on dynamic Source removal** #995

all fixes will be made on this branch.

rjhowell44, 2023-08-22:
Looks to be similar to #984

rjhowell44, 2023-08-23:
@githubzhao735 I've pretty much convinced myself that this is a know issue with NVIDIA's old Streammuxer.  I first logged the issue back in Sept 2020... see https://forums.developer.nvidia.com/t/releasing-nvstreammux-request-pad-results-in-a-deadlock/155271

In a previous DSL release, I used to check the URI using opencv first before connecting it to the Streammuxer.  I had to remove this check when NVIDIA deprecated the use of opencv and made it very difficult to support. This is probably why we are seeing the error again.

Anyway, NVIDIA has been working on a new streammuxer for sometime now and will become the default Streammuxer in the next DeepStream release.   We can use the Streammuxer now by setting an appropriate build flag. See docs here https://docs.nvidia.com/metropolis/deepstream/dev-guide/text/DS_plugin_gst-nvstreammux2.html#

My plan is the release v0.27.alpha this week... then make upgrading to the new Streammuxer the focus of the next release.  

Once complete we can verify that this issue has been solved.

githubzhao735, 2023-08-27:
I'm looking forward to the new branch solving this problem. Thanks very much!

hadestyz, 2024-01-11:
@rjhowell44 It seems that this issue is still unresolved on version 6.3 of Deepstream?

rjhowell44, 2024-01-15:
@hadestyz a very good question... I know that NVIDIA has continued to fix issues with the old Streammux plugin in DS 6.3 and 6.4.  They are just not very forthcoming when it comes to disclosure of fixes.

My suggestion would be to try a re-verify with v0.28.alpha and the old Streammuxer. For any problems rediscovered,  we can open new issues with NVIDIA. 

Also, will be interesting to see what the come back with on my other issues like: https://forums.developer.nvidia.com/t/pipelines-with-new-nvstreammux-and-nvstreamdemux-fail-to-play-correctly-in-ds-6-3/278396

rjhowell44, 2024-03-23:
I'm unable to reproduce this issue in DS 6.4 if after performing the following. From the DeepStream getting started

_Sometimes with RTSP streams the application gets stuck on reaching EOS. This is because of an issue in rtpjitterbuffer component. To fix this issue,a script “update_rtpmanager.sh” at /opt/nvidia/deepstream/deepstream/ has been provided with required details to update gstrtpmanager library. The script should be executed once above mentioned packages are installed as prerequisite._

closed 2024-03-23 by rjhowell44