prominenceai/deepstream-services-library #1056: How to configure the config_infer_secondary_xxx.txt correctly when adding a Secondary classification Inference?

MintyLee, 2023-08-28:
When I do detection + classification model inference, I want to apply the classification model to each object(My initial purpose was to crop out the input image according to the bbox, and feed the cropped raw image data into the classification model).
the preprocessing method when my classification model is trained is:
transforms.Compose([transforms.Resize(input_size), transforms.ToTensor(), transforms.Normalize((0.485, 0.456, 0.406), (0.229, 0.224, 0.225))])(img),and the model input size is n*3*256*128(NCHW)，model-color-format=RGB.
so,The config file(Reference config_infer_secondary_carcolor.txt) for my second stage model is:
gpu-id=0
net-scale-factor=0.01735207
model-engine-file=../../xxx.engine
parse-classifier-func-name=parse_bbox_softmax
labelfile-path=../../xx.txt
batch-size=1
model-color-format=0
network-input-order=0
process-mode=2
offsets=124;117;104
network-mode=2
is-classifier=1
output-blob-names=softmax
classifier-async-mode=1
classifier-threshold=0.7
input-object-min-width=40
input-object-min-height=75
operate-on-gie-id=1
operate-on-class-ids=0
classifier-type=reflect
scaling-filter=1
maintain-aspect-ratio=0
symmetric-padding=0

But the classification model output is inconsistent with my own implementation of pytorch inference,Please help see where there is a bug, thank you very much, please!!!

rjhowell44, 2023-08-31:
@MintyLee Sorry, but I'm not the export on this.  My expertise is more in gstreamer and deepstream deployment.  Have you asked your question on the NVIDIA boards?  https://forums.developer.nvidia.com/c/accelerated-computing/intelligent-video-analytics/deepstream-sdk/15

 You could also try on our discord side... https://discord.gg/MJvY9jjpAK

rjhowell44, 2023-09-26:
Sorry I was unable to help with this.

closed 2023-09-26 by rjhowell44