prominenceai/deepstream-services-library #1110: RTSP connection issue

youngjae-avikus, 2023-11-10:
hello
I am testing with a new RTSP camera, which has the feature of requiring id/passwd. (like `std::wstring rtsp_camera_path = L"rtsp://admin:1234@192.168.0.41:554/Ch1";`

But in conclusion, connection does not work in DSL. h264 depay and parse work, but it seems like something isn't supported behind the scenes.

DS: 6.2
commit: v0.27.b.

```bash
previous state = DSL_STATE_PAUSED, new state = DSL_STATE_PLAYING
0:00:02.419810207  9396 0x56374f118b00 INFO                     DSL DslPipelineStateMgr.cpp:339:HandleBusWatchMessage:  : Message type : progress
0:00:02.419857199  9396 0x56374f118b00 INFO                     DSL DslPipelineStateMgr.cpp:340:HandleBusWatchMessage:  :    source    : rtsp-source-rtspsrc
0:00:02.419869119  9396 0x56374f118b00 INFO                     DSL DslPipelineStateMgr.cpp:341:HandleBusWatchMessage:  :    type      : 1
0:00:02.419874825  9396 0x56374f118b00 INFO                     DSL DslPipelineStateMgr.cpp:342:HandleBusWatchMessage:  :    code      : open
0:00:02.419882225  9396 0x56374f118b00 INFO                     DSL DslPipelineStateMgr.cpp:343:HandleBusWatchMessage:  :    text      : Retrieving server options
0:00:02.458234482  9396 0x56374f118b00 INFO                     DSL DslPipelineStateMgr.cpp:339:HandleBusWatchMessage:  : Message type : progress
0:00:02.458265482  9396 0x56374f118b00 INFO                     DSL DslPipelineStateMgr.cpp:340:HandleBusWatchMessage:  :    source    : rtsp-source-rtspsrc
0:00:02.458279781  9396 0x56374f118b00 INFO                     DSL DslPipelineStateMgr.cpp:341:HandleBusWatchMessage:  :    type      : 1
0:00:02.458290465  9396 0x56374f118b00 INFO                     DSL DslPipelineStateMgr.cpp:342:HandleBusWatchMessage:  :    code      : open
0:00:02.458298675  9396 0x56374f118b00 INFO                     DSL DslPipelineStateMgr.cpp:343:HandleBusWatchMessage:  :    text      : Retrieving media info
0:00:02.464212408  9396 0x56374f314d80 INFO                     DSL DslSourceBintr.cpp:3546:HandleSelectStream:  : Media = 'video' for RtspSourceBitnr 'rtsp-source'
0:00:02.464238760  9396 0x56374f314d80 INFO                     DSL DslSourceBintr.cpp:3548:HandleSelectStream:  : Encoding = 'H264' for RtspSourceBitnr 'rtsp-source'
0:00:02.468330741  9396 0x56374f314d80 WARN                  udpsrc gstudpsrc.c:1445:gst_udpsrc_open:<udpsrc0> warning: Could not create a buffer of requested 524288 bytes (Operation not permitted). Need net.admin privilege?
0:00:02.468366409  9396 0x56374f314d80 WARN                  udpsrc gstudpsrc.c:1455:gst_udpsrc_open:<udpsrc0> have udp buffer of 212992 bytes while 524288 were requested
0:00:02.468532963  9396 0x56374f118b00 INFO                     DSL DslPipelineStateMgr.cpp:339:HandleBusWatchMessage:  : Message type : progress
0:00:02.468558392  9396 0x56374f118b00 INFO                     DSL DslPipelineStateMgr.cpp:340:HandleBusWatchMessage:  :    source    : rtsp-source-rtspsrc
0:00:02.468570119  9396 0x56374f118b00 INFO                     DSL DslPipelineStateMgr.cpp:341:HandleBusWatchMessage:  :    type      : 1
0:00:02.468580390  9396 0x56374f118b00 INFO                     DSL DslPipelineStateMgr.cpp:342:HandleBusWatchMessage:  :    code      : request
0:00:02.468590023  9396 0x56374f118b00 INFO                     DSL DslPipelineStateMgr.cpp:343:HandleBusWatchMessage:  :    text      : SETUP stream 0
terminate called after throwing an instance of 'std::logic_error'
  what():  basic_string::_M_construct null not valid
Aborted (core dumped)
```

**_result of gdb_**

```bash
(gdb) bt
#0  0x00007fd7b2f1300b in raise () at /lib/x86_64-linux-gnu/libc.so.6
#1  0x00007fd7b2ef2859 in abort () at /lib/x86_64-linux-gnu/libc.so.6
#2  0x00007fd7b32cc8d1 in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007fd7b32d837c in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007fd7b32d83e7 in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007fd7b32d8699 in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007fd7b32cf1dc in std::__throw_logic_error(char const*) () at /lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00007fd7c2024044 in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string<std::allocator<char> >(char const*, std::allocator<char> const&) [clone .constprop.0] () at /workspace/FaceUS/install/lib/libdsl.so
#8  0x00007fd7c204376d in DSL::RtspSourceBintr::HandleSelectStream(_GstElement*, unsigned int, _GstCaps*) () at /workspace/FaceUS/install/lib/libdsl.so
#9  0x00007fd7c2044d4b in DSL::RtspSourceSelectStreamCB(_GstElement*, unsigned int, _GstCaps*, void*) () at /workspace/FaceUS/install/lib/libdsl.so
#10 0x00007fd79fee6ff5 in  () at /lib/x86_64-linux-gnu/libffi.so.7
#11 0x00007fd79fee640a in  () at /lib/x86_64-linux-gnu/libffi.so.7
#12 0x00007fd7b548a30d in g_cclosure_marshal_generic () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#13 0x00007fd7b5489802 in g_closure_invoke () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#14 0x00007fd7b549d814 in  () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#15 0x00007fd7b54a847d in g_signal_emit_valist () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#16 0x00007fd7b54a90f3 in g_signal_emit () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#17 0x00007fd71c4e593f in  () at /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstrtsp.so
#18 0x00007fd71c4e989e in  () at /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstrtsp.so
#19 0x00007fd71c4ee610 in  () at /usr/lib/x86_64-linux-gnu/gstreamer-1.0/libgstrtsp.so
#20 0x00007fd7bd6381e7 in  () at /lib/x86_64-linux-gnu/libgstreamer-1.0.so.0
#21 0x00007fd7c19f2374 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#22 0x00007fd7c19f1ad1 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#23 0x00007fd7c195b609 in start_thread () at /lib/x86_64-linux-gnu/libpthread.so.0
#24 0x00007fd7b2fef133 in clone () at /lib/x86_64-linux-gnu/libc.so.6
```

rjhowell44, 2023-11-10:
@youngjae-avikus When you are entering your credentials as  `username@password`, is there any chance that either the username or password has any special characters in it, such as ( / ? @ %) ?

rjhowell44, 2023-11-10:
I've done a lot of searching on this... there were a number of posts on the NVIDIA boards, but with no answers.  this was the only hint I could find which is why I'm asking about the special chars https://github.com/Kurento/bugtracker/issues/554

youngjae-avikus, 2023-11-10:
> ungjae-avikus When you are entering your credentials as `username@password`, is there any chance that either the username or password has any special characters in it, such as ( / ? @ %) ?

No. I used admin / 1234

rjhowell44, 2023-11-11:
That should be ok... it is just if you have @ in the password or user id. 

There are a number of threads on this subject, none that offer much help. For example, see https://github.com/bluenviron/mediamtx/issues/1019

There is this thread on NVIDIA boards that offers some insight https://forums.developer.nvidia.com/t/gst-pipeline-using-rtspsrc-and-nvv4l2decoder-doesnt-work-in-deepstream-6-1-1-anymore/233760/4

Could you try capturing a pipeline graph?  I'd like to see the connection between the `rtpsrc->parser->depay->decoder`

youngjae-avikus, 2023-11-11:
> That should be ok... it is just if you have @ in the password or user id.
> 
> There are a number of threads on this subject, none that offer much help. For example, see [bluenviron/mediamtx#1019](https://github.com/bluenviron/mediamtx/issues/1019)
> 
> There is this thread on NVIDIA boards that offers some insight https://forums.developer.nvidia.com/t/gst-pipeline-using-rtspsrc-and-nvv4l2decoder-doesnt-work-in-deepstream-6-1-1-anymore/233760/4
> 
> Could you try capturing a pipeline graph? I'd like to see the connection between the `rtpsrc->parser->depay->decoder`

Well, actually, the deepstream version that contains the face recognition module I developed is 6.1.1. I think it's worth upgrading the version and testing it out. The other module I tested separately was 6.2 (which worked somewhat normally but low performance). I'll take some time to test it and share it with you.

youngjae-avikus, 2023-11-12:
@rjhowell44 

deepstream 6.2
jetson orin agx devkit
code: example/cpp/rtsp_connection.cpp

Still getting the same error

youngjae-avikus, 2023-11-12:
[image]
@rjhowell44 
Here it is mydot

youngjae-avikus, 2023-11-12:
When analyzing issues on the forum, it seems there is something to do with `nvv4l2decoder and pasebin`.
It doesn't seem to have anything to do with id/passwd.

closed 2024-02-02 by youngjae-avikus