prominenceai/deepstream-services-library #1116: When using RTSP source, on-screen-display and window-sink operation stops. (not sync)

youngjae-avikus, 2023-11-20:
Code content: 
Code that receives pgie as tensor-meta and draws bbox on osd (later Add align - sgie process). In order to `get_nvds_buf_surface` in the code, the **buffer must be RGBA, so set the source_output_buffer_format** like this.

problem:
In case of file-source, osd-sink is updated, but in case of rtsp-source, only the first frame appears and the buffer is not synchronized thereafter. However, when I recorded the FPS, the current 15 FPS was displayed well. In other words, there is no major problem, but it seems that sync is not working well on the rendering side.

```python
#!/usr/bin/env python

import os, sys
import io
import yaml

sys.path.append('../dsl')
from dsl import *
import pyds
    
# 전역 변수로 설정
global config

def load_config():
    global config
    with open('config_face_recognize.yml', 'r') as file:
        config = yaml.safe_load(file)
        
def get_label_names_from_file(filepath):
    """ Read a label file and convert it to string list """
    f = io.open(filepath, "r")
    labels = f.readlines()
    labels = [elm[:-1] for elm in labels]
    f.close()
    return labels

def add_obj_meta_to_frame(frame_object, batch_meta, frame_meta, label_names, unique_id, scale_x, scale_y, maintain_aspect_ratio, left_padding, top_padding):
    obj_meta = pyds.nvds_acquire_obj_meta_from_pool(batch_meta)
        
    # Aspect ratio 조정
    if maintain_aspect_ratio:
        scale = min(scale_x, scale_y)
        scale_x, scale_y = scale, scale
    else:
        scale_x, scale_y = scale_x, scale_y
    
    rect_params = obj_meta.rect_params
    rect_params.left = (frame_object.left - left_padding) * scale_x
    rect_params.top = (frame_object.top - top_padding) * scale_y
    rect_params.width = frame_object.width * scale_x
    rect_params.height = frame_object.height * scale_y
    
    obj_meta.confidence = frame_object.detectionConfidence
    obj_meta.class_id = frame_object.classId
    
    # Semi-transparent yellow background
    rect_params.has_bg_color = 0
    rect_params.bg_color.set(1, 1, 0, 0.4)

    # Red border of width 5
    rect_params.border_width = 5
    rect_params.border_color.set(1, 0, 0, 1)
    
    obj_meta.object_id = config['primary_gie']['untracked_object_id']
    obj_meta.unique_component_id = unique_id
    
    lbl_id = frame_object.classId
    if lbl_id >= len(label_names):
        raise ValueError("Label ID {} is out of range".format(lbl_id))
        
    # Set the object classification label.
    obj_meta.obj_label = label_names[lbl_id]
    
[... 287 lines cut ...]
        if retval != DSL_RETURN_SUCCESS:
            break
        retval = dsl_pipeline_eos_listener_add('pipeline', eos_event_listener, None)
        if retval != DSL_RETURN_SUCCESS:
            break

        # Play the pipeline
        retval = dsl_pipeline_play('pipeline')
        if retval != DSL_RETURN_SUCCESS:
            break

        dsl_main_loop_run()
        retval = DSL_RETURN_SUCCESS
        break

    # Print out the final result
    print(dsl_return_value_to_string(retval))

    dsl_pipeline_delete_all()
    dsl_component_delete_all()

if __name__ == '__main__':
    load_config()
    sys.exit(main(sys.argv))
```

youngjae-avikus, 2023-11-20:
I found a solution, but it seems **async has become false with the recent dsl update.** But could it be a problem in the future whether it works or not depending on async? As far as I know, it is related to sgie, but I am trying to connect sgie separately, and window-sink is also needed.

```python
retval = dsl_sink_async_enabled_set('window-sink', True)
        if retval != DSL_RETURN_SUCCESS:
            break
```

HoangTienDuc, 2023-11-23:
Hi @youngjae-avikus, Have you tried to convert RGBA frame to get RGB frame?
```
           try:
                frame_meta = pyds.NvDsFrameMeta.cast(l_frame.data)

            except StopIteration:
                break
            n_frame = pyds.get_nvds_buf_surface(gst_buffer, frame_meta.batch_id)
            in_surf_height, in_surf_width = n_frame.shape[:2]
            frame_copy = np.array(n_frame, copy=True, order='C')
            rgb_frame = cv2.cvtColor(frame_copy, cv2.COLOR_RGBA2BGR)
            print(n_frame.shape)
```
I tried it in different ways, but it is failed. If you are success with it, can you pls share it with me?

closed 2024-02-02 by youngjae-avikus