prominenceai/deepstream-services-library #1132: How to dynamically add source at 0.28

hadestyz, 2024-01-09:
def _CreatePerSourceComponents(self, pipeline, source_id, uri, ode_handler):
        # New Component names based on unique source name
        components = ComponentNames(source_id)

        # For each camera, create a new RTSP Source for the specific RTSP URI
        if self.drop_frame_interval == 1:
            drop_frame_interval = 0
        else:
            drop_frame_interval = self.drop_frame_interval

        if uri.find("rtsp") != -1:
            retval = dsl_source_rtsp_new(components.source,
                                         uri=uri,
                                         protocol=DSL_RTP_TCP,
                                         skip_frames=0,
                                         drop_frame_interval=drop_frame_interval, latency=100, timeout=2)
        else:
            retval = dsl_source_uri_new(components.source, uri, False, False, drop_frame_interval)

        if (retval != DSL_RETURN_SUCCESS):
            return retval

        retval = dsl_source_video_buffer_out_format_set(components.source, DSL_VIDEO_FORMAT_RGBA)
        if retval != DSL_RETURN_SUCCESS:
            return retval

        retval = dsl_pph_buffer_timeout_new('{}-timeout-handler'.format(components.source), self.source_timeout, self._timeout_callback, components)
        if retval != DSL_RETURN_SUCCESS:
            return retval

        retval = dsl_source_pph_add(components.source, '{}-timeout-handler'.format(components.source))
        if retval != DSL_RETURN_SUCCESS:
            return retval
        return retval

    def run(self):
        while True:
           #Add first source
            retval = self._CreatePerSourceComponents('pipeline', 1, self.video_url, None)
            if retval != DSL_RETURN_SUCCESS:
                break

            # New Primary GIE using the filespecs above, with infer interval
            retval = dsl_infer_gie_primary_new('primary-gie', self.inferConfigFile, self.modelEngineFile, 0)
            if retval != DSL_RETURN_SUCCESS:
                break

            # New IOU Tracker, setting operational width and hieght
            retval = dsl_tracker_new('iou-tracker', self.iou_tracker_config_file, 480, 272)
            if retval != DSL_RETURN_SUCCESS:
                break

            # New Tiler, setting width and height, use default cols/rows set by source count
            retval = dsl_tiler_new('tiler', TILER_WIDTH, TILER_HEIGHT)
            if retval != DSL_RETURN_SUCCESS:
                break

            retval = dsl_pph_custom_new('custom-pph',client_handler=self._save_frame, client_data=None)
            if retval != DSL_RETURN_SUCCESS:
                break

            retval = dsl_pph_custom_new('custom-pph2',client_handler=self._tiler_sink_pad_buffer_probe, client_data=None)
            if retval != DSL_RETURN_SUCCESS:
                break

            # Add the ODE Pad Probe Hanlder to the Sink Pad of the Tiler
            retval = dsl_tiler_pph_add('tiler', 'custom-pph', DSL_PAD_SINK)
            if retval != DSL_RETURN_SUCCESS:
                break

            retval = dsl_tiler_pph_add('tiler', 'custom-pph2', DSL_PAD_SINK)
            if retval != DSL_RETURN_SUCCESS:
                break

            retval = dsl_sink_fake_new('fake-sink')
            if retval != DSL_RETURN_SUCCESS:
                break

            # Add all the components to our pipeline
            retval = dsl_pipeline_new_component_add_many('pipeline',['source-1', 'primary-gie', 'iou-tracker', 'tiler', 'fake-sink', None])
            if retval != DSL_RETURN_SUCCESS:
                break

            #0.28-removed
            """
            retval = dsl_pipeline_streammux_batch_properties_set('pipeline', self.MAX_SOURCE_COUNT, 40000)
            # retval = dsl_pipeline_streammux_batch_properties_set('pipeline', 1, 40000)
            if retval != DSL_RETURN_SUCCESS:
                break
            retval = dsl_pipeline_streammux_dimensions_set('pipeline', self.max_width, self.max_height)
            if retval != DSL_RETURN_SUCCESS:
                break
            """

            #0.28-add
            retval = dsl_pipeline_streammux_batch_size_set('pipeline', 16)
            # retval = dsl_pipeline_streammux_batch_properties_set('pipeline', 1, 40000)
            if retval != DSL_RETURN_SUCCESS:
                break

            retval = dsl_pipeline_eos_listener_add('pipeline', self._eos_event_listener, None)
            if retval != DSL_RETURN_SUCCESS:
                break

            # Play the pipeline
            retval = dsl_pipeline_play('pipeline')
            if retval != DSL_RETURN_SUCCESS:
                break

            dsl_main_loop_run()
            retval = DSL_RETURN_SUCCESS
            break

        # Print out the final result
        print('pipeline end：',dsl_return_value_to_string(retval))

        dsl_pipeline_delete_all()
        dsl_component_delete_all()
        
After completing the assembly of the pipeline, I dynamically added a source and encountered the following error:

0:00:18.886760352 55136 0x7f5ea800c580 WARN          nvvideoconvert gstnvvideoconvert.c:1957:gst_nvvideoconvert_fixate_caps:<source-2-nvvideoconvert-buffer-out> nvbuf-memory-type property is set based on SRC caps. Property config setting (if any) is overridden!!
0:00:18.886806559 55136 0x7f5ea800c580 WARN          nvvideoconvert gstnvvideoconvert.c:1962:gst_nvvideoconvert_fixate_caps:<source-2-nvvideoconvert-buffer-out> gpu-id property is set based on SRC caps. Property config setting (if any) is overridden!!
[NvTiler::ScaleAndPlaceAllMeta] ERROR: 175; scaleX=0.110937 scaleY=0.000000 frameMeta=0x7f5ef402e100
[NvTiler::Composite] ERROR: 343; ScaleAndPlaceAllMeta failed (0x7f5f9c02b7c0;0x7f5f9c03ec20)

The inference for the first source was successfully completed, but there were issues with the dynamically added source.

By the way, it seems that the parameters here are incorrect

[image]

rjhowell44, 2024-01-09:
@hadestyz unfortunately, I've found a number issues with the new Streammux. The issue above included. 

I'm going to need to reenable use of the old Streammux and make the new new Streammux optional... something I was hoping to avoid, but it's clear that the new plugin is not ready for all cases. 

This will take me a bit of time, but I'll let you know when available.

rjhowell44, 2024-01-15:
@hadestyz 
The old Streammux services have be re-enabled, verified, and - merged into the `v0.28.alpha` branch

And thanks for reporting the bug above.  New issue opened for this see #1134 - also merged into the `v0.28.alpha` branch