prominenceai/deepstream-services-library #1165: How can I use gstreamer plugin in dsl which is not part of deepstream?

navyverma, 2024-03-14:


rjhowell44, 2024-03-14:
@navyverma which gstreamer plugin are you referring to?

navyverma, 2024-03-14:
This is our custom gstreamer based plugin

rjhowell44, 2024-03-14:
Interesting... can I ask where in the pipeline you add your plugin, and does it have gst-properties that can be set/get?

navyverma, 2024-03-15:
Mutiple video sources -> streaamux -> infer -> tracking -> demuxer ->  custom_plugin ->osd -> msg_broker
Mutiple video sources ->                                                                       ->  custom_plugin ->osd -> msg_broker
Mutiple video sources ->                                                                       ->  custom_plugin ->osd -> msg_broker
Mutiple video sources ->                                                                       ->  custom_plugin ->osd -> msg_broker

And yes it has some gst-properties that can be set.

rjhowell44, 2024-03-21:
@navyverma my apologies for the late response.  DSL does not currently support this, but I can see the value in having it. I will try and schedule this for the next release.

rjhowell44, 2024-03-25:
@navyverma, I've started working on a solution for this... a couple of questions.
1.  does your plugin support signals?
2.  what language do you prefer/use for your application development, C/C++ or Python?

navyverma, 2024-03-25:
@rjhowell44 Yes, plugin does suppoort gstreamer's signal, and we are using c++ to build plugin

rjhowell44, 2024-03-26:
@navyverma New issue to track this #1174

rjhowell44, 2024-04-04:
This also required #1176.  The implementation and test-coverage for both issues are complete. Documentation and examples are still to be done.  I should have this finished by early next week.

rjhowell44, 2024-04-16:
#1176 and #1174 have been completed and merged into the `v0.30.alpha` dev branch.  The formal release date for `v0.30.alpha` is TBD.  

@navyverma if you're still interested in this, it is now ready for trial.  Feedback is always greatly appreciated. Let me know if there is anything missing for your use case.

navyverma, 2024-04-17:
@rjhowell44 Thanks. I really appreciate your work. Will try and test within a few days and share feedback with you.

closed 2024-04-26 by navyverma