prominenceai/deepstream-services-library #1182: Segmentation fault when set pipeline with RTSP source created from rtsp-simple-server

subzeromot, 2024-04-20:
Currently, I'm using rtsp-simple-server to create a rtsp stream from video
- run simple rtsp server docker:
	`docker run --rm -it -e RTSP_PROTOCOLS=tcp -p 8554:8554 aler9/rtsp-simple-server`
- stream video to rtsp server:
	`ffmpeg -re -stream_loop -1 -i <video_path> -f rtsp rtsp://localhost:8554/mystream`
- play:
	`ffplay rtsp://localhost:8554/mystream`

I can use this RTSP stream with `deepstream-app `and `deepstream-python-app` as well.
But when I use dsl to create rtsp sink, the system return: Segmentation fault (core dumped). I used `dsl_source_rtsp_new` to created rtsp sink.
Here is my code:

```python

import sys
import time
from dsl import *

# update host URL to IP address if clients are off device.
# host_uri = '0.0.0.0'

file_path = "/opt/nvidia/deepstream/deepstream/samples/streams/sample_1080p_h265.mp4"
rtsp_url = "rtsp://192.168.100.13:8554/mystream"

# Filespecs (Jetson and dGPU) for the Primary GIE
primary_infer_config_file = \
    '/opt/nvidia/deepstream/deepstream/samples/configs/deepstream-app/config_infer_primary.txt'
primary_model_engine_file = \
    '/opt/nvidia/deepstream/deepstream/samples/models/Primary_Detector/resnet10.caffemodel_b8_gpu0_int8.engine'

# Filespec for the IOU Tracker config file
iou_tracker_config_file = \
    '/opt/nvidia/deepstream/deepstream/samples/configs/deepstream-app/config_tracker_IOU.yml'

## 
# Function to be called on XWindow KeyRelease event
## 
def xwindow_key_event_handler(key_string, client_data):
    print('key released = ', key_string)
    if key_string.upper() == 'P':
        dsl_pipeline_pause('pipeline')
    elif key_string.upper() == 'R':
        dsl_pipeline_play('pipeline')
    elif key_string.upper() == 'Q' or key_string == '' or key_string == '':
        dsl_pipeline_stop('pipeline')
        dsl_main_loop_quit()
 
## 
# Function to be called on XWindow Delete event
## 
def xwindow_delete_event_handler(client_data):
    print('delete window event')
    dsl_pipeline_stop('pipeline')
    dsl_main_loop_quit()

## 
# Function to be called on End-of-Stream (EOS) event
## 
def eos_event_listener(client_data):
    print('Pipeline EOS event')
    dsl_pipeline_stop('pipeline')
    dsl_main_loop_quit()

## 
# Function to be called on every change of Pipeline state
## 
def state_change_listener(old_state, new_state, client_data):
    print('previous state = ', old_state, ', new state = ', new_state)
    if new_state == DSL_STATE_PLAYING:
        dsl_pipeline_dump_to_dot('pipeline', "state-playing")

def main(args):

    # Since we're not using args, we can Let DSL initialize GST on first call
    while True:

        # New File Source using the file path specified above, repeat enabled.
        # retval = dsl_source_file_new('file-source', file_path, True)
        # retval = dsl_source_uri_new('file-source', file_path, is_live=True, skip_frames=0, drop_frame_interval=0)
        retval = dsl_source_rtsp_new(name='file-source', uri=rtsp_url, protocol=DSL_RTP_ALL, skip_frames=0, latency=1000, timeout=0, drop_frame_interval=0)
        
        if retval != DSL_RETURN_SUCCESS:
            break

        # New Primary GIE using the filespecs above with interval = 0
        retval = dsl_infer_gie_primary_new('primary-gie', 
            primary_infer_config_file, primary_model_engine_file, 0)
        if retval != DSL_RETURN_SUCCESS:
            break

        # New IOU Tracker, setting operational width and height.
        retval = dsl_tracker_new('iou-tracker', iou_tracker_config_file, 480, 272)
        if retval != DSL_RETURN_SUCCESS:
            break

        # New OSD with text, clock and bbox display all enabled. 
        retval = dsl_osd_new('on-screen-display', 
            text_enabled=True, clock_enabled=True, 
            bbox_enabled=True, mask_enabled=False)
        if retval != DSL_RETURN_SUCCESS:
            break

        # New EGL Window Sink with offests and dimensions
        retval = dsl_sink_window_egl_new('window-sink', 100, 100, 1280, 720)
        if retval != DSL_RETURN_SUCCESS:
            break
        # Add the XWindow event handler functions defined above
        retval = dsl_sink_window_key_event_handler_add('window-sink', 
            xwindow_key_event_handler, None)
        if retval != DSL_RETURN_SUCCESS:
            break
        retval = dsl_sink_window_delete_event_handler_add('window-sink', 
            xwindow_delete_event_handler, None)
        if retval != DSL_RETURN_SUCCESS:
            break

        # Add all the components to our pipeline
        retval = dsl_pipeline_new_component_add_many('pipeline', 
            ['file-source', 'primary-gie', 'iou-tracker', 'on-screen-display', 
            'window-sink', None])
        if retval != DSL_RETURN_SUCCESS:
            break

        ## Add the listener callback functions defined above
        retval = dsl_pipeline_state_change_listener_add('pipeline', 
            state_change_listener, None)
        if retval != DSL_RETURN_SUCCESS:
            break
        retval = dsl_pipeline_eos_listener_add('pipeline', 
            eos_event_listener, None)
        if retval != DSL_RETURN_SUCCESS:
            break

        # Play the pipeline
        retval = dsl_pipeline_play('pipeline')
        if retval != DSL_RETURN_SUCCESS:
            break

        dsl_main_loop_run()
        retval = DSL_RETURN_SUCCESS
        break

    # Print out the final result
    print(dsl_return_value_to_string(retval))

    dsl_delete_all()

if __name__ == '__main__':
    sys.exit(main(sys.argv))

```

subzeromot, 2024-04-20:
I'm using docker image `nvcr.io/nvidia/deepstream:6.2-devel` and run with dgpu

rjhowell44, 2024-04-21:
@subzeromot thanks for reporting this. 

Could you look in your DSL Makefile and please tell me what version you are using?

and could you provide me with a log file?
```
export GST_DEBUG_FILE=./log.txt
export GST_DEBUG=1,DSL:4
```
So far, I'm unable to reproduce any issue with the cameras that I have.  Would you be willing to debug this issue for me a little? Do you use vscode?

rjhowell44, 2024-04-23:
@subzeromot I'm unable to reproduce any issue... If I try with you commands above, I see
```
INFO                     DSL src/DslSourceBintr.cpp:3579:HandleSelectStream:  : Media = 'video' for RtspSourceBitnr 'file-source'
INFO                     DSL src/DslSourceBintr.cpp:3581:HandleSelectStream:  : Encoding = 'MP4V-ES' for RtspSourceBitnr 'file-source'
ERROR                    DSL src/DslSourceBintr.cpp:3633:HandleSelectStream:  : Unsupported encoding = 'MP4V-ES' for RtspSourceBitnr 'file-source'
ERROR                rtspsrc gstrtspsrc.c:7966:gst_rtspsrc_open_from_sdp:<file-source-rtspsrc> setup failed
ERROR                    DSL src/DslPipelineStateMgr.cpp:472:HandleErrorMessage:  : Error message 'No supported stream was found. You might need to allow more transport protocols or may otherwise be missing the right GStreamer RTSP extension plugin.' received from 'file-source-rtspsrc'
```
...but no crash.

If I update the ffmpeg command to include `-vcodec h264` your script above works correct for me.

subzeromot, 2024-04-24:
> @subzeromot I'm unable to reproduce any issue... If I try with you commands above, I see
>
> ```
> INFO                     DSL src/DslSourceBintr.cpp:3579:HandleSelectStream:  : Media = 'video' for [...]

thanks, it worked now

closed 2024-04-24 by subzeromot