prominenceai/deepstream-services-library #1186: Tracking algorithm for all Triggers is incorrect/insufficient for how the Tracker libs actually work.

rjhowell44, 2024-04-21:
The current Tracking algorithm(s) delete a tracked object once it is no longer found in frame, however, the Tracker libs are capable or reacquiring objects once reported by the inference engine again.  The DSL tracking needs to be able to support the same.

rjhowell44, 2024-04-22:
Tracker config files specify two config properties relavant to this.
```yml
  maxShadowTrackingAge: 51    # Max length of shadow tracking. If the shadowTrackingAge exceeds this limit, the tracker will be terminated.
  earlyTerminationAge: 1    # If the shadowTrackingAge reaches this threshold while in TENTATIVE period, the target will be terminated prematurely.
```
**NVIDIA Sample Configs:**
* IOU - `maxShadowTrackingAge: 38`
* NvDCF_Acuracy - `maxShadowTrackingAge: 42`
* NvDCF_Perf - `maxShadowTrackingAge: 51`
* NvDeepSORT - `maxShadowTrackingAge: 68` 

<br> 

There will be a new DSL symbolic constant defined as
```c
/**
 * @brief The maximum number of consecutive frames a tracked object
 * can go undetected before it is purged and no longer tracked. 
*/
#define DSL_ODE_TRACKED_OBJECT_MISSING_FROM_FRAME_MAX               100
```

rjhowell44, 2024-04-22:
merged into the `v0.30.alpha` branch

rjhowell44, 2024-05-29:
Although the above change will improve the behavior, there is a fundamental difference in the way the Tracker and ODE Trigger work.
- Tracker maintains up to a fixed number of undetected objects indefinitely.
- ODE Trigger maintains undetected objects for a max number of frames.

The DSL Tracking need to be changed to match the Tracker

Opi-Nguyen, 2024-12-12:
> Tracker config files specify two config properties relavant to this.
> 
> ```yaml
>   maxShadowTrackingAge: 51    # Max length of shadow tracking. If the shadowTrackingAge exceeds this limit, the tracker will be terminated.
>   earlyTerminationAge: 1    # If the shadowTrackingAge reaches this threshold while in TENTATIVE period, the target will be terminated prematurely.
> ```
> 
> **NVIDIA Sample Configs:**
> 
> * IOU - `maxShadowTrackingAge: 38`
> * NvDCF_Acuracy - `maxShadowTrackingAge: 42`
> * NvDCF_Perf - `maxShadowTrackingAge: 51`
> * NvDeepSORT - `maxShadowTrackingAge: 68`
> 
> 
> There will be a new DSL symbolic constant defined as
> 
> ```c
> /**
>  * @brief The maximum number of consecutive frames a tracked object
>  * can go undetected before it is purged and no longer tracked. 
> */
> #define DSL_ODE_TRACKED_OBJECT_MISSING_FROM_FRAME_MAX               100
> ```

so, how can i config tracking old ? by maxShadowTrackingAge as Nvidia config file or by DSL_ODE_TRACKED_OBJECT_MISSING_FROM_FRAME_MAX. If using DSL_ODE_TRACKED_OBJECT_MISSING_FROM_FRAME_MAX how can i config that, let show me an example

rjhowell44, 2024-12-12:
@Opi-Nguyen my apologies if the Issue above is unclear or misleading.   The DSL tracking logic will work correctly as long as `DSL_ODE_TRACKED_OBJECT_MISSING_FROM_FRAME_MAX` is greater than the  Tracker's `maxShadowTrackingAge`. 

I plan to add a separate API to make `DSL_ODE_TRACKED_OBJECT_MISSING_FROM_FRAME_MAX` updatable in the future.  Is this something you think need now?

Opi-Nguyen, 2024-12-15:
> @Opi-Nguyen my apologies if the Issue above is unclear or misleading. The DSL tracking logic will work correctly as long as `DSL_ODE_TRACKED_OBJECT_MISSING_FROM_FRAME_MAX` is greater than the [...]

Yes, thanks for your information!

rjhowell44, 2024-12-16:
I will try and add the services in the up coming release.  In the meantime, you can simply update the `DSL_ODE_TRACKED_OBJECT_MISSING_FROM_FRAME_MAX ` in `DslApi.h` and rebuild DSL if needed.