prominenceai/deepstream-services-library #1197: Send client data to callback function

subzeromot, 2024-04-29:
In `smart_record_tap_start_session_on_user_demand.py` how can I send more data to the callback function `record_event_listener`? I saw `client_data`, I tried to parse my data to this but it seems not work.

rjhowell44, 2024-04-29:
@subzeromot please see the [smart_record_tap_start_session_on_ode_occurrence.py](https://github.com/prominenceai/deepstream-services-library/blob/master/examples/python/smart_record_tap_start_session_on_ode_occurrence.py) for an example of how to work with `client_data`

The class that is used to define the data is here https://github.com/prominenceai/deepstream-services-library/blob/9e5b49cf6960edaca97ffaeb9fcb22cbaaaaadf0/examples/python/smart_record_tap_start_session_on_ode_occurrence.py#L171

how to provide `client_data` here https://github.com/prominenceai/deepstream-services-library/blob/9e5b49cf6960edaca97ffaeb9fcb22cbaaaaadf0/examples/python/smart_record_tap_start_session_on_ode_occurrence.py#L293

how to access `client_data` in callback here https://github.com/prominenceai/deepstream-services-library/blob/9e5b49cf6960edaca97ffaeb9fcb22cbaaaaadf0/examples/python/smart_record_tap_start_session_on_ode_occurrence.py#L189

subzeromot, 2024-04-30:
thanks, it worked

closed 2024-04-30 by subzeromot