prominenceai/deepstream-services-library #1206: DSL src/DslPipelineStateMgr.cpp:472:HandleErrorMessage:  : Error message 'Internal data stream error.' received from 'tiler-queue'

hadestyz, 2024-05-09:
Hi :
Deepstream 6.4
dsl 0.29
cuda 12,2
driver 535.104.12

When I initialized the pipeline and tried to add a new source to it, the analysis and inference seemed to have stopped. 

  ```
  def run(self):
        # Since we're not using args, we can Let DSL initialize GST on first call
        while True:
            retval = self._CreatePerSourceComponents('pipeline', 1, self.video_url, None)
            if retval != DSL_RETURN_SUCCESS:
                break

            # New Primary GIE using the filespecs above, with infer interval
            retval = dsl_infer_gie_primary_new('primary-gie', self.inferConfigFile, self.modelEngineFile, 0)
            if retval != DSL_RETURN_SUCCESS:
                break

            # New IOU Tracker, setting operational width and hieght
            retval = dsl_tracker_new('iou-tracker', self.iou_tracker_config_file, 480, 272)
            if retval != DSL_RETURN_SUCCESS:
                break

            # New Tiler, setting width and height, use default cols/rows set by source count
            retval = dsl_tiler_new('tiler', TILER_WIDTH, TILER_HEIGHT)
            if retval != DSL_RETURN_SUCCESS:
                break

            retval = dsl_pph_custom_new('custom-pph',client_handler=self._save_frame, client_data=None)
            if retval != DSL_RETURN_SUCCESS:
                break

            retval = dsl_pph_custom_new('custom-pph2',client_handler=self._tiler_sink_pad_buffer_probe, client_data=None)
            if retval != DSL_RETURN_SUCCESS:
                break

            # Add the ODE Pad Probe Hanlder to the Sink Pad of the Tiler
            retval = dsl_tiler_pph_add('tiler', 'custom-pph', DSL_PAD_SINK)
            if retval != DSL_RETURN_SUCCESS:
                break

            retval = dsl_tiler_pph_add('tiler', 'custom-pph2', DSL_PAD_SINK)
            if retval != DSL_RETURN_SUCCESS:
                break

            retval = dsl_sink_fake_new('fake-sink')
            if retval != DSL_RETURN_SUCCESS:
                break

            # Add all the components to our pipeline
            retval = dsl_pipeline_new_component_add_many('pipeline',['source-1', 'primary-gie', 'iou-tracker', 'tiler', 'fake-sink', None])
            if retval != DSL_RETURN_SUCCESS:
                break

            ### IMPORTANT: we need to explicitely set the stream-muxer Batch properties, otherwise the Pipeline
            # will use the current number of Sources when set to Playing, which would be 1 and too small
            retval = dsl_pipeline_streammux_batch_properties_set('pipeline', self.MAX_SOURCE_COUNT, 40000)
            # retval = dsl_pipeline_streammux_batch_properties_set('pipeline', 1, 40000)
            if retval != DSL_RETURN_SUCCESS:
                break
            retval = dsl_pipeline_streammux_dimensions_set('pipeline', self.max_width, self.max_height)
            if retval != DSL_RETURN_SUCCESS:
                break

            retval = dsl_pipeline_eos_listener_add('pipeline', self._eos_event_listener, None)

            if retval != DSL_RETURN_SUCCESS:
[... 169 lines cut ...]
                                        float(obj_meta.detector_bbox_info.org_bbox_coords.top) + float(obj_meta.detector_bbox_info.org_bbox_coords.height),
                                        float(obj_meta.confidence), obj_meta.class_id])
                    try:
                        l_obj = l_obj.next
                    except StopIteration:
                        print('_tiler_sink_pad_buffer_probe error3 --------------------------------')
                        break
                if track_boxes and source_id in self.stream_data.keys():

                    _, match_ids = self.stream_data[source_id]["byte_track"].update(np.array(track_boxes))
                    for _, track in enumerate(match_ids):
                        boxes[int(track[-1])].append(int(track[0]))

            if source_id != 0 and frame_number * self.drop_frame_interval % self.frame_skip == 0 and source_id in self.stream_data.keys() and self.stream_data[source_id]["task_id"] in self.img_list_dict.keys():
           pass

            try:
                l_frame = l_frame.next
            except StopIteration:
                print('_tiler_sink_pad_buffer_probe error4 --------------------------------')
                break

        return DSL_PAD_PROBE_OK
```
Do you have any ideas? Thank you

rjhowell44, 2024-05-10:
@hadestyz I'll be able to look at this tomorrow

hadestyz, 2024-05-10:
@rjhowell44 I compared the examples and found the cause of the problem:
If I use this, it will result in the above error: retval=dsl_source_uri.new (components. source, uri, False, False, 4)
If I use this, it looks like it runs normally: retval=dsl_source_uri.new (components. source, uri, False, 0, 0)