prominenceai/deepstream-services-library #1224: how to pMapperedBuffer to opencv Mat?

kimsjpk1, 2024-05-31:
i tried to apply super resolution in deepstream code

and i watched your github code

example/cpp/4file_custom_pph_using_opencv.cpp

// Transforming only one frame in the batch, so create a copy of the single 
            // surface ... becoming our new source surface. This creates a new mono 
            // (non-batched) surface copied from the "batched frames" using the batch id 
            // as the index
            DSL::DslMonoSurface monoSurface(pMappedBuffer->pSurface, 
                pFrameMeta->batch_id);

            // Coordinates and dimensions for our destination surface.
            int left(0), top(0);

            // capturing full frame
            int width = pMappedBuffer->GetWidth(pFrameMeta->batch_id);
            int height = pMappedBuffer->GetHeight(pFrameMeta->batch_id);

if pMappedBuffer with batch_id convert to opencv Mat, i can apply super resolution on Frame

please tell me how to convert it

thanks in advance

kimsjpk1, 2024-05-31:
i search information o found it

https://forums.developer.nvidia.com/t/deepstream-sample-code-snippet/142683

but i want to change buffer before applied objected detection 

in deepstream-service-library/src/DslSurfaceTransform.h, i want to change code

DslMappedBuffer(GstBuffer* pBuffer)
            : GstMapInfo GST_MAP_INFO_INIT
            , m_pBuffer(NULL)
        {
            LOG_FUNC();
            
            if (!gst_buffer_map(pBuffer, this, GST_MAP_READ))
            {
                LOG_ERROR("Failed to map gst buffer");
                throw;
            }
            // set only when successful
            m_pBuffer = pBuffer;

            pSurface = (NvBufSurface*)data;
        }

pSurface->surfaceList[index] to cv::Mat and apply super resolution to pSurface

rjhowell44, 2024-06-03:
@kimsjpk1 I'll take a look at this tomorrow (Monday)

rjhowell44, 2024-06-05:
@rjhowell44 my apologies for the slow response.  After some review, I think I understand what you're looking for. First, DslSurfaceTransform.h was developed to simplify the process of copying a frame to a new buffer surface so it can be mapped and manipulated. 

If I understand correctly, you want to update the actual pipeline buffer prior to your inference component. 

Looking at the code snippets you provided, this should be possible if done in a custom pph added to the sink pad of the primary inference component.   You won't need DslSurfaceTransform.h.

The opencv super resolution looks really interesting, and I can see the value in having an example of this.  I'm currently working on #1225 and should be done in a couple of days.  If you can give me a few days, I should be able to work on this next.