prominenceai/deepstream-services-library #1236: dsl_remuxer_branch_remove

shancaidazf, 2024-07-01:
the method dsl_remuxer_branch_remove  seems to make no sense

rjhowell44, 2024-07-01:
@shancaidazf DSL supports the user's ability to completely assemble and disassemble Pipelines at runtime.  For, every `Add` service there will corresponding `Remove` service.  Just as for every `New` service (constructor) there will be a corresponding `Delete` service (destructor).

shancaidazf, 2024-07-02:
> @shancaidazf DSL supports the user's ability to completely assemble and disassemble Pipelines at runtime. For, every `Add` service there will corresponding `Remove` service. Just as for every `New` service (constructor) there will be a corresponding `Delete` service (destructor).

sorry,  i mean the method dsl_remuxer_branch_remove is supposed to remove the branch ,but it seems to do nothing

rjhowell44, 2024-07-02:
There are a number of static and dynamic test cases for the Demuxer... all tests are passing. What makes you think it isn't working? 
* Please describe how to reproduce.  
* What version of DeepStream are you using?
* What version of DSL are you using?
* Are you using the old or new streammuxer?
* please provide a log for the failing case

```
export GST_DEBUG=1,DSL:4
```

shancaidazf, 2024-07-02:
it is not demuxer.  i refered to remuxer.    i just modifyed the script parallel_inference_on_selective_streams.py with a xwindow_key_event_handler and execute  dsl_remuxer_branch_remove('remuxer','mybranch').  But it actually do nothing.

rjhowell44, 2024-07-03:
Sorry for misreading your questions.   The Remuxer does not support dynamic branch add/remove while the pipeline is playing. The Remuxer must be in an unlinked idle state before the branch can be removed. If you call `dsl_remuxer_branch_remove` while the pipeline is playing the service will return `DSL_RESULT_REMUXER_BRANCH_REMOVE_FAILED`.

Is this what you are trying to do.