prominenceai/deepstream-services-library #1263: Issue with video recording duration in DeepStream SDK

rafico, 2024-08-16:
Hello,
I'm working on recording videos using the 'smart_record_tap_start_session_on_user_demand.py' example from the DeepStream SDK. I've encountered an issue where the recording consistently stops after exactly 30 seconds, regardless of the settings I use.
I've attempted to modify the 'start-time' and 'duration' arguments passed to the 'dsl_tap_record_session_start' function, but these changes do not affect the recording duration. The behavior remains unchanged.
Given this consistent 30-second limit despite parameter adjustments, I suspect there might be a bug in the 'dsl_tap_record_session_start' function or in its interaction with other components.
Could you please investigate this issue or provide guidance on how to correctly set the recording duration?
Thank you for your assistance.
Best regards,

Rafi

rjhowell44, 2024-08-16:
@rafico what version of DeepStream are you using?  And what version of DSL are you on?

rafico, 2024-08-16:
Hi,
I'm using DS version 7 and the DSL v0.30.a.alpha.

rjhowell44, 2024-08-16:
@rafico what size are you setting the cache with dsl_sink_record_cache_size_set ?

rafico, 2024-08-16:
Hi,
I set the cache to 300 seconds (5 minuts) with dsl_tap_record_cache_size_set.

I used the following sequence of commands:

# returns the default 30 seconds
retval, cache_size = dsl_tap_record_cache_size_get('my-record-tap')
# return 0 - success
retval = dsl_tap_record_cache_size_set('my-record-tap', 300)
# returns 300 seconds - just to verify the cache was changed
retval, cache_size = dsl_tap_record_cache_size_get('my-record-tap')

The recording still stops after 30 seconds.

rjhowell44, 2024-08-16:
@rafico ok, thanks for the info... I'll see if I can reproduce this today

rjhowell44, 2024-08-17:
@rafico I'm able to reproduce this.  It's either a change in DS behavior, or a misunderstanding on my part? Either way, it's any easy issue to fix

Using the debugger... 

If we look at the structure used to create the smart recording context -- at the time the Pipeline is linked and set to a state of playing -- there is a field that defines the `defaultDuration` for all recordings that is set to 30s

[image]

Even though the cache size is set to 400s and a value of 300s is used for the duration...  

[image]

It appears the the `defaultDuration` is acting as a **_`maximim-duration`_**

The workaround is to set the default-duration in `DslApi.h` to something large enough for your use case. Search for.
```c
#define DSL_DEFAULT_VIDEO_RECORD_DURATION_IN_SEC                    30
```

I need to think about how best to handle this in DSL. I will probable set the value to something large, and then add services to allow the client to update it. The fact that default == max will need to be covered in the docs.

rafico82, 2024-08-18:
Hi,
The workaround was successful. Thank you!

rjhowell44, 2024-08-18:
See [dsl_pipeline_streammux_sync_inputs_enabled_set](https://github.com/prominenceai/deepstream-services-library/blob/master/docs/api-pipeline.md#dsl_pipeline_streammux_sync_inputs_enabled_set), but keep in mind that this is synchronization by the Streammuxer. The example above uses the Recording-Tap component which taps into the Source pre decoder which is upstream of the streammuxer. If synchronization is important, you may want look at using a demuxer with multiple Recording Sinks... downstream of the streammuxer.

closed 2024-08-24 by rjhowell44