prominenceai/deepstream-services-library #1271: 'GstNvArgusCameraSrc' has no property named 'bufapi-version'

yliuosu, 2024-08-30:
I followed the instructions to compile and build the dsl library successfully. But when I run the test code

#!/usr/bin/env python
from dsl import *

# New CSI Live Camera Source
retval = dsl_source_csi_new('csi-source', 1280, 720, 30, 1)

if retval != DSL_RETURN_SUCCESS:
    print(retval)
    # --- handle error
    
I got the error information: (python:330436): GLib-GObject-CRITICAL **: 22:57:55.694: g_object_set_is_valid_property: object class 'GstNvArgusCameraSrc' has no property named 'bufapi-version'

I don't know what caused this problem. Anyone knows how to solve it. Thanks.
[image]
[image]
[image]

yliuosu, 2024-08-31:
Thank you so much for your prompt and detailed response!

rjhowell44, 2024-08-31:
@yliuosu I'm going to keep this issue open until I have it resolved.

yliuosu, 2024-09-03:
Hi, Sir,

I had another question about how to view the RTSP Server output . I saw the RTSP Server Sink example code:

retval = dsl_sink_rtsp_server_new('rtsp-sink', 
            host = "0.0.0.0",       # 0.0.0.0 = "this host, this network."
            udp_port = 5400,        # UDP port 5400 uses the Datagram Protocol.             
            rtsp_port = 8554,       # 
            codec = DSL_CODEC_H265, # High Efficiency Video Coding (HEVC)
            bitrate = 0,            # Set to 0 to use plugin default (4000000)
            interval = 0)  
            
how can I view the rtsp stream in the vlc player, I tired "rtsp://localhost:8554", "rtsp://0.0.0.0:8554/", "rtsp://localhost:8554/ds-test",  "rtsp://0.0.0.0:8554/ds-test" in vlc player . None of them worked and I didn't find any example about how to view the rtsp output.  

Thanks for your time

rjhowell44, 2024-09-03:
@yliuosu I switched from VLC to [Pot Player](https://potplayer.daum.net/) some time ago. I find it much easier to test with. You might consider doing the same.

Once installed, go to  `Open->Open URL`

[image]

The URL is composed of 
```
rtsp://<device-name>:<password>@<ip-address>:8554/<sink-name>
```
for example, if my case I have
device-name = `"prominenceai-dgpu-01"`
password = `"123456"`
ip-address of device = `"192.168.1.31"`
sink-name = `"rtsp-sink"`

which gives me a url of 
`rtsp://prominenceai-dgpu-01:123456@192.168.1.31:8554/rtsp-sink`

Also, please be aware that there are currently 3 open issues related to Encode Sinks (RTSP Sink being one of them)
* #1273 
* #1274 
* #1272

I've recently completed this work and have pushed the changes into the `[v0.01.beta](https://github.com/prominenceai/deepstream-services-library/tree/v0.01.beta) development branch.  

I plan to release this with in the next couple of days (although it may be released as `v0.31.alpha`)

You may want to consider moving to this branch now.  There are a couple of breaking changes... i.e. the encoder constant values to include software encoding and parameter name changes. 

Let me know if you're still having problems. 
Robert.

rjhowell44, 2024-09-03:
@yliuosu I took a quick look at the RTSP Sink documentation and found that the description of the URL was wrong.  I've #1281 for this.

yliuosu, 2024-09-04:
Thank you. This has been very helpful.

closed 2024-09-05 by rjhowell44