prominenceai/deepstream-services-library #1311: Doubt regarding paralle inference pipeline

shreejalt, 2024-11-23:
Hi all,
Thanks a lot for your contribution to this great repository.
I had one doubt regarding the parallel inference pipeline.

I have one source and one PGIE. I want to pass this PGIE output to two different trackers running in parallel. So, this one source will have two different outputs i.e. one from tracker 1 and oter from tracker 2. 

I want to then combine the metadata using something like metamux and then get the output in two different streams.

How can I do this using your repository.

Please if someone can help me regarding this.

THanks,
Shreejal

rjhowell44, 2024-11-25:
@shreejalt I've been trying to think of a way of implementing the case you've described...  kind of parallel tracking with a single inference engine. 

If you look at the image below

[image]

The key to parallel inference is that each branch starts with its own streammuxer 
[image]

which creates/adds new metadata for the branch.  If we remove the inference engines from the branches and just added trackers, and then add a single engine upstream before the branches are created, the streammuxers for each branch will strip/drop any frame/object metadata before adding the new metadata. The metadata from the inference engine will never get to the Trackers. 

Have you considered just adding two pgies using the same model, one to each branch?  If both models are producing roughly the same metadata, for each branch, will this work for you? The metamuxer will combine the metadata from both branches.

shreejalt, 2024-11-25:
Hi @rjhowell44 

Thank your for your response. 
i think it will be very redundant for me to do that, because my PGIE gives output at the same confidence score, and I want you to get the tracking results at different tracking thresholds for the same output. 

I can definitely add the two same PGIEs before the tracking after the branch happens. But will it affect the performance of the system?

Also, is there no other alternative to pass the metadata if we keep the pgie upstream? Can we force pass the metadata like generating new NvDSFrameData before passing it to the trackers?

Please let me know your thoughts!

rjhowell44, 2024-11-25:
@shreejalt it looks like I might be wrong, and that the Streammuxers can cascade the data from upstream.  I need to look into this in more detail... see https://docs.nvidia.com/metropolis/deepstream/dev-guide/text/DS_plugin_gst-nvstreammux2.html#cascaded-muxing

I'm in the middle of wrapping up some work.  If you can give me a day, I'll to do some more research later and see if I can get this to work.

shreejalt, 2024-11-25:
@rjhowell44 Thanks a lot. Sure no worries. If you can help me in this it would be greatly appreciated. Even the architecture diagram would work after you test it in your system. 

Awaiting your response! I can wait until till week

rjhowell44, 2024-11-29:
@shreejalt my apologies for delay. Busy week....

First, I needed to create a new repo branch (candidate patch release) to allow a Tracker to be added as a "complete" Remuxer inference branch. See [v0.31.c.alpha](https://github.com/prominenceai/deepstream-services-library/tree/v0.31.c.alpha) 

And here's a python script I've created to work on this.  [parallel_tracking_with_single_pgie.py.txt](https://github.com/user-attachments/files/17962004/parallel_tracking_with_single_pgie.py.txt). You will need to strip off the `.txt.` extension after downloading. 

The example uses a single PGIE with one IOU Tracker and one DCF Tracker. An ODE Occurrence Trigger is used to trigger on every occurrence of a bicycle object, with a ODE Print Action to print out each detected object's metadata.

A couple of observations.

1. There is currently no way of determining which Tracker/branch updated the metadata and added the Tracking ID. See the Object metadata structure below. The only way you can tell in my example is that the IOU Tracker sets the `tracker_confidence=-1`. 
 
```c
/**
 * Holds metadata for an object in the frame.
 */
typedef struct _NvDsObjectMeta {
  NvDsBaseMeta base_meta;
  /** Holds a pointer to the parent @ref NvDsObjectMeta. Set to NULL if
   no parent exists. */
  struct _NvDsObjectMeta *parent;
  /** Holds a unique component ID that identifies the metadata
   in this structure. */
  gint unique_component_id;
  /** Holds the index of the object class inferred by the primary
   detector/classifier. */
  gint class_id;
  /** Holds a unique ID for tracking the object. @ref UNTRACKED_OBJECT_ID
   indicates that the object has not been tracked. */
  guint64 object_id;
  /** Holds a structure containing bounding box parameters of the object when
    detected by detector. */
  NvDsComp_BboxInfo detector_bbox_info;
  /** Holds a structure containing bounding box coordinates of the object when
   * processed by tracker. */
  NvDsComp_BboxInfo tracker_bbox_info;
  /** Holds a confidence value for the object, set by the inference
   component. confidence will be set to -0.1, if "Group Rectangles" mode of
   clustering is chosen since the algorithm does not preserve confidence
   values. Also, for objects found by tracker and not inference component,
   confidence will be set to -0.1 */
  gfloat confidence;
  /** Holds a confidence value for the object set by nvdcf_tracker.
   * tracker_confidence will be set to -0.1 for KLT and IOU tracker */
  gfloat tracker_confidence;
  /** Holds a structure containing positional parameters of the object
   * processed by the last component that updates it in the pipeline.
   * e.g. If the tracker component is after the detector component in the
   * pipeline then positinal parameters are from tracker component.
   * Positional parameters are clipped so that they do not fall outside frame
   * boundary. Can also be used to overlay borders or semi-transparent boxes on
   * objects. @see NvOSD_RectParams. */
  NvOSD_RectParams rect_params;
  /** Holds mask parameters for the object. This mask is overlayed on object
   * @see NvOSD_MaskParams. */
  NvOSD_MaskParams mask_params;
  /** Holds text describing the object. This text can be overlayed on the
   standard text that identifies the object. @see NvOSD_TextParams. */
  NvOSD_TextParams text_params;
  /** Holds a string describing the class of the detected object. */
  gchar obj_label[MAX_LABEL_SIZE];
  /** Holds a pointer to a list of pointers of type @ref NvDsClassifierMeta. */
  NvDsClassifierMetaList *classifier_meta_list;
  /** Holds a pointer to a list of pointers of type @ref NvDsUserMeta. */
  NvDsUserMetaList *obj_user_meta_list;
  /** Holds additional user-defined object information. */
  gint64 misc_obj_info[MAX_USER_FIELDS];
  /** For internal use. */
  gint64 reserved[MAX_RESERVED_FIELDS];
}NvDsObjectMeta;
```

2. I never see objects from the same frame tracked by both trackers. I'm not sure if the nvmetamuxer failing to mux correctly, or if this is the nature of the two trackers and their config parameters.  It will need to be looked at in more detail.

Opi-Nguyen, 2024-12-15:
> @shreejalt I've been trying to think of a way of implementing the case you've described... kind of parallel tracking with a single inference engine.
> 
> If you look at the image below
> 
> ![image](https://private-user-images.githubusercontent.com/7819483/389582279-8772a4ef-2dd7-47e7-a55e-acd00b8d3935.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQyODQxMjEsIm5iZiI6MTczNDI4MzgyMSwicGF0aCI6Ii83ODE5NDgzLzM4OTU4MjI3OS04NzcyYTRlZi0yZGQ3LTQ3ZTctYTU1ZS1hY2QwMGI4ZDM5MzUucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIxNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEyMTVUMTczMDIxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZTQ2MzAwMTVkNjIyZDIzMDUzMTBjYzkzNGZmOGRlZTFjMzYyNTgwN2U3NDdhZjRkMzY0ODFmOGVhZjRiZjcxZiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.O0YCCGuSzWZOe6Ln-Xw77lI447uNw7WVAwklX2RMyqk)
> 
> The key to parallel inference is that each branch starts with its own streammuxer ![streammux](https://private-user-images.githubusercontent.com/7819483/389589319-3581bed3-9939-4ac1-8636-6b17720aba2f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQyODQxMjEsIm5iZiI6MTczNDI4MzgyMSwicGF0aCI6Ii83ODE5NDgzLzM4OTU4OTMxOS0zNTgxYmVkMy05OTM5LTRhYzEtODYzNi02YjE3NzIwYWJhMmYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIxNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEyMTVUMTczMDIxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ODM1ZGFjN2JmNDViNDlhNDlkNDFlN2JmOWE1MWMwMjZlNWJjYWQ2YmRkNTgzZDJhYzQ1NWUwYjZiNWEwYjkzNyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.3fcn54QAmeLZYFrCvu-53vr0M_UyECyAzErCef8zy7o)
> 
> which creates/adds new metadata for the branch. If we remove the inference engines from the branches and just added trackers, and then add a single engine upstream before the branches are created, the streammuxers for each branch will strip/drop any frame/object metadata before adding the new metadata. The metadata from the inference engine will never get to the Trackers.
> 
> Have you considered just adding two pgies using the same model, one to each branch? If both models are producing roughly the same metadata, for each branch, will this work for you? The metamuxer will combine the metadata from both branches.

I have tried your pipeline on 'parallel_inference_on_selective_streams.py', and the model engines in the branch are generated with a batch size equal to the number of streams the branch needs to process. This poses a challenge if I already have a pre-generated engine. For example, an engine generated with a batch size of 2 is sufficient for 2 RTSP streams, but if I dynamically add a branch or add more RTSP sources while running, it will fail.

By the way, can you provide me with an example of adding an RTSP source or adding a branch at runtime? I currently need to add/delete sources or branches during runtime.

rjhowell44, 2024-12-16:
@Opi-Nguyen unfortunately, I have found that NVIDIA's metamux plugin does not support dynamic stream updates.  I tried to bring this up with NVIDIA here, https://forums.developer.nvidia.com/t/questions-about-the-deepstream-metamuxer-plugin/297668... but they don't seem that interested in supporting this.  

Just a comment about dynamic source updates in general (not supported with parallel inference).  The batch-size used to generate the model, and the batch-size of the pipeline streammux must be equal to or greater-than the maximum number of sources that will be added at one time.