prominenceai/deepstream-services-library #1318: Custom Object Tracking

Opi-Nguyen, 2024-12-11:
i use yolox, is there a way to only include the bbox of class 0 (person) in the tracker?
And how can the incremented ID variables only apply to class 0 (person)?

rjhowell44, 2024-12-12:
@Opi-Nguyen if you are using a Primary GIE you should be able to set the `operate-on-class-ids` property in your inference config  file.  See https://docs.nvidia.com/metropolis/deepstream/dev-guide/text/DS_plugin_gst-nvinfer.html#gst-properties

Try
`operate-on-class-id = 0`

Opi-Nguyen, 2024-12-12:
> @Opi-Nguyen if you are using a Primary GIE you should be able to set the `operate-on-class-ids` property in your inference config file. See https://docs.nvidia.com/metropolis/deepstream/dev-guide/text/DS_plugin_gst-nvinfer.html#gst-properties
> 
> Try `operate-on-class-id = 0`

`operate-on-class-ids` is the key in the group [property], it has an effect on secondary-gie
I tried to set `filter-out-class-ids` to operate only on `class = 0` by this way:
`onnx-file=/app/models/yolox/yolox.onnx
model-engine-file=/app/models/yolox/model_b1_gpu0_fp32.engine
#int8-calib-file=calib.table
labelfile-path=labels.txt
batch-size=1
network-mode=0
num-detected-classes=80
filter-out-class-ids = 1;2;3;4;5;6;7;8;9;10;11;12;13;14;15;16;17;18;19;20;21;22;23;24;25;26;27;28;29;30;31;32;33;34;35;36;37;38;39;40;41;42;43;44;45;46;47;48;49;50;51;52;53;54;55;56;57;58;59;60;61;62;63;64;65;66;67;68;69;70;71;72;73;74;75;76;77;78;79
interval=0
gie-unique-id=1
process-mode=1`
It is working on Pgie

closed 2024-12-23 by Opi-Nguyen