prominenceai/deepstream-services-library #1330: Error with gst_queue_leak_downstream and Aborted (core dumped)

LocNgoXuan23, 2025-01-20:
Hi @rjhowell44 
I really appreciate your GitHub repository, and I have been using it for some time. It has been extremely important for my major project.
Currently, I am facing a significant issue, and I need your help.

Sometimes, the system runs perfectly fine for a while, but then I encounter this error, and I have no idea what is causing it:
```
**                                                       
ERROR:../plugins/elements/gstqueue.c:1147:gst_queue_leak_downstream: assertion failed: (leak != NULL)              
Bail out! ERROR:../plugins/elements/gstqueue.c:1147:gst_queue_leak_downstream: assertion failed: (leak != NULL)
Aborted (core dumped)
```

Here is my env:
```
Deepstream version: 7.0
CPU: Intel Xeon Processor (Icelake) 16 cores
RAM: 64GB
GPU: Nvidia A30 (24 GB VRam)
```

This is the detail log:
```
FPS 0 (AVG)    FPS 1 (AVG)    FPS 2 (AVG)    FPS 3 (AVG)                                                                                                                                                                    [168/1890]
0.00 (14.98)   14.90 (17.11)  15.10 (15.26)  15.05 (15.26)                                                                                                                                                                            
                                                                                                                                                                                                                                      
                                                                                                                                                                                                                                      
 ***  Recording Event  ***                                                                                                                                                                                                            
session_id:  0                                                                                                                                                                                                                        
event:       DSL_RECORDING_EVENT_END                                                                                                                                                                                                  
filename:    4-record-tap_00052_20250120-101328_738496.mp4                                                                                                                                                                            
dirpath:     /root/deploy/media                                                                                                                                                                                                       
duration:    35883                                                                                                                                                                                                                    
container:   0                                                                                                                                                                                                                        
width:       1920                                                                                                                                                                                                                     
height:      1080                                                                                                                                                                                                                     
                                                                                                                                                                                                                                      
FPS 0 (AVG)    FPS 1 (AVG)    FPS 2 (AVG)    FPS 3 (AVG)                                                                                                                                                                              
0.00 (14.98)   14.90 (17.11)  14.71 (15.26)  14.95 (15.26)                                                                                                                                                                            
                                                                                                                                                                                                                                      
[PGIE] RTSP Source previous state =  4 , new state =  1                                                                                                                                                                               
[PGIE] RTSP Source previous state =  1 , new state =  4                                                                                                                                                                               
[PGIE] RTSP Source previous state =  4 , new state =  1                                                                                                                                                                               
                                                                                                                   
FPS 0 (AVG)    FPS 1 (AVG)    FPS 2 (AVG)    FPS 3 (AVG)                                                           
0.00 (14.98)   15.06 (17.10)  15.26 (15.26)  15.06 (15.26)                                                                                                                                                                            
                                                                                                                                                                                                                                      
[PGIE] RTSP Source previous state =  1 , new state =  4  
                                                                                                                                                                                                                                      
FPS 0 (AVG)    FPS 1 (AVG)    FPS 2 (AVG)    FPS 3 (AVG)                                                           
0.00 (14.98)   15.12 (17.10)  14.99 (15.26)  14.99 (15.26)                                                                                                                                                                            
                                                                                                                   
[PGIE] RTSP Source previous state =  4 , new state =  1                                                                                                                                                                               
[PGIE] RTSP Source previous state =  1 , new state =  4                                                            
                                                                                                                                                                                                                                      
FPS 0 (AVG)    FPS 1 (AVG)    FPS 2 (AVG)    FPS 3 (AVG)                                                           
0.00 (14.98)   14.88 (17.10)  15.01 (15.26)  15.01 (15.26)                                                                                                                                                                            
                                                                                                                                                                                                                                      
[PGIE] RTSP Source previous state =  4 , new state =  1                                                                                                                                                                               
[PGIE] RTSP Source previous state =  1 , new state =  4
                                                         
[... cut ...]
0.00 (14.98)   15.02 (17.09)  15.02 (15.26)  15.02 (15.26)                                                         
                                                                                                                   
**                                                       
ERROR:../plugins/elements/gstqueue.c:1147:gst_queue_leak_downstream: assertion failed: (leak != NULL)              
Bail out! ERROR:../plugins/elements/gstqueue.c:1147:gst_queue_leak_downstream: assertion failed: (leak != NULL)
Aborted (core dumped)
```

rjhowell44, 2025-01-21:
@LocNgoXuan23 I'm not familiar with this error.  I did a quick search and found a couple of topics on the NVIDIA dev boards

https://forums.developer.nvidia.com/t/error-gstqueue-cgst-queue-leak-downstream-assertion-failed-leak-null/290424

and 

https://forums.developer.nvidia.com/t/memory-leak-in-test5-app-with-smart-record/191121

However, neither were answered.  Could you capture some more complete logs for me?  First, you can direct the log output to file with
```
export GST_DEBUG_FILE=./output.txt
```
Then set the level first with

```
export GST_DEBUG=1,DSL:4
```
and a second file with 
```
export GST_DEBUG=4,DSL:0
```
Can you also look at your DSL Makefile and let me know what version of DSL your are using.... thanks.

LocNgoXuan23, 2025-01-21:
Hi @rjhowell44 , tks for your answer
I have been monitoring since I created this issue and haven't encountered this error again.

Additionally, I have just run the following three commands to monitor the bug:
```
export GST_DEBUG_FILE=./output.txt
export GST_DEBUG=1,DSL:4
export GST_DEBUG=4,DSL:0
```

And this my makefile. 
[README.md](https://github.com/user-attachments/files/18486379/README.md)

closed 2025-02-26 by LocNgoXuan23